open-ephys/plugin-GUI #151: Urgent: "Found corrupted record" error

shayo, 2017-07-31:
I keep getting the error "Found corrupted record" when trying to parse a continuous data stream.
Looking at the samples that are being read using "load_open_ephys_data_faster", I see
  1024
1024
1024.....
And then a bunch of crap like this:
       64260
       27653
       44805
       57605
       
I suspect this is from pausing and resuming the recording.

This happens with the latest version of the master branch.

Aaron, have you experienced this in the past? Did you test pausing/resume recently?

shayo, 2017-07-31:
Confirmed. This does not happen if I don't do "pause" and resume.

aacuevas, 2017-07-31:
Weird, I'm testing this at this very moment and I cannot reproduce the error. Is there a particular signal chain you're using? How high is the cpu usage meter?

shayo, 2017-07-31:
cpu is very low.
I am using my new spike sorting module, but it will be strange if that is the reason because there isn't anything funky going on there...
Let me test with the vanilla master branch. I'll update soon.

shayo, 2017-07-31:
Confirmed. This happens with the vanilla master branch.
Signal chain:
RHYTHM->Bandpass->SpikeSorter->LFP viewer.

CPU usage never passed 5%.

aacuevas, 2017-07-31:
Still can't reproduce it...
Does it happen in all channels or just in some?
Can you please make a brief (a couple of seconds) recording with a pause and resume that gets you corrupted saves, and send it to me (the whole folder) so I can take a look at them and see if I can find anything out?

shayo, 2017-07-31:
Just sent it to you over email.
I just discovered that it only happens for the ADC channels.
i.e., within the same recording session, CH1 is fine, but ADC1 is not.

aacuevas, 2017-07-31:
I managed to reproduce the issue. It's not necessarily ADC channels, but it happens when non-contiguous channels are recorded. For example, saving just CH1 and ADC1 makes ADC1 corrupted, but saving all but CH1 makes CH2 corrupted and everything else fine while saving just ADC1-8 makes most of them corrupt. I still have to find a reliable failure pattern, but this seems a serious bug indeed. I'll need a bit of time to track it, though.

In the meanwhile, viable workarounds are either saving all the channels, which seems to work fine, or using the "Use separate files" record option for the open ephys format (accesible via the "R" button next to the record format selector), which will create different files for each recording.

Also, other recording formats seem unaffected, since they all create separate files/datasets for each recording.

shayo, 2017-07-31:
OK. Keep me posted. It's good to know you managed to reproduce this. I
wonder how no one came across this until now...

P.S.
When you do get this fixed, please remember another old request I had - to
indicate the recording session on the GUI somewhere, so you'll know which
one it is. Thanks!

aacuevas, 2017-08-02:
I've pushed a fix that seems to solve the issue, at least in the test I've run. Can you pelase check if this solves this (and #154 ) for you? If so I'll update the binaries with the fixed code.

shayo, 2017-08-02:
Yes. I'll test this first thing tomorrow morning.

shayo, 2017-08-02:
Great work @aacuevas . This solved both problems I reported (corrupted files and timestamp mismatch)

closed 2017-08-02 by shayo