open-ephys/plugin-GUI #153: LFP Viewer drawing bug

shayo, 2017-07-31:
I'm still experiencing drawing issues with LFP Viewer (TTL events are not drawn correctly):
[image]

jvoigts, 2017-07-31:
this is fairly old code so i don't think it will make a difference, but just to be sure, could you try the lfp viewer beta?

shayo, 2017-07-31:
Looks slightly better, but still same issue.
[image]

pbaker2, 2018-02-21:
We are having a similar issue, after about 3 minutes of recording, our events get really messy on the LFP viewer. Additionally, it will occasionally start leaving an event channel in the high position (when using a gate) which is really bad when using the arduino output to a laser. I'm guessing that it has something to do with the event channels backing up in a buffer or something but the CPU meter on the GUI isn't really suggesting that it should be bogging down. I'm no expert in the data stream though so I may be missing something. Any insight into this?

closed 2020-12-15 by medengineer