open-ephys/plugin-GUI #154: Urgent: incorrect timestamps when reading ADC continuous channel

shayo, 2017-07-31:
I've done a short test recording of a photodiode signal (to make sure things are properly synchronized in my rig). 
I observed weird and non-expected behavior when looking at the timestamps returned from load_open_ephys_data_faster.

When parsing the continuous file, nsamples are all 1024. Plotting  info.ts looks good on the surface:

[image]

But when I look more carefully (plot diff), I get the following:
[image]

This doesn't look right.
Also, when I plot the signal, aligned to these timestamps, it artificially stretches some intervals:
[image]

I can hookup a signal generator, but I'm pretty sure I'll also see a skewed sine wave.
For reference, here is the same data, plotted without the timestamps (i.e., assuming fixed sampling rate):

[image]

This data was from an ADC channel.
Looking at another channel (ch1 - spike), diff returns a perfect 1024.

This may or may not be related to the other issue I found today. As a side note - this recorded session was not paused or resumed.

shayo, 2017-08-01:
FYI: @aacuevas , @jsiegle , @jvoigts 
This goes away when I record all channels, so this is probably related to the other bug that I reported.
I'm guessing the index increment is incorrect in RHD2000Thread::updateBuffer() ?!?!?

aacuevas, 2017-08-01:
I'm inclined to think the issue is in the recording part, since the number of recorded channels is irrelevant to the hardware buffers. But it might very well be related

shayo, 2017-08-02:
Recent patch to OriginalRecording solved the issue.

closed 2017-08-02 by shayo