open-ephys/plugin-GUI #155: CoreServices::getTimestamp() issue

shayo, 2017-08-01:
Aaron,I have an issue with CoreServices::getTimestamp()

I would like to be able to call it at any given moment, and get a reliable estimate of the current hardware timestamp.  However, at the moment, getTimestamp is only updated every buffer read event (i.e.,~33ms).

For spike sorting module, precise hardware timestamp was done by adding the sample number within the buffer. This works great for processors that analyze the buffer in the "process" function, but some processors (i.e. Network events, or the new spike sorting module I'm working on) handle things in separate threads, and events can occur in them at any moment.  Thus, when I call to CoreServices::getTimestamp(), I have inaccurate timestamp (up to 33 ms).

What I would like to have is something like this:
Whenever a hardware timestamp arrives (t_h), a software timestamp is appended to it (t_s).
When a timestamp request is made, a software timestamp (t_r) is taken.
Then, the precise hardware timestamp is calculated as follows: (t_r-t_s) * sample_rate + t_h

Can we add this as another core service function? (maybe call is CoreServices::getEstimatedTimestamp), or any other name you think is more suitable.

shayo, 2017-08-01:
Alternatively, can we append this directly to the TIMESTAMP event:

	```

	int64 ts_software = CoreServices::getSoftwareTimestamp();
	uint8 data_software_timestamp[8];
	memcpy(data_software_timestamp, &ts_software, 8);

    // generate timestamp
    addEvent(events,    // MidiBuffer
             TIMESTAMP, // eventType
             0,         // sampleNum
             nodeId,    // eventID
             0,      // eventChannel
             16,         // numBytes
             data,   // data
             true    // isTimestampEvent
            );

```

aacuevas, 2017-08-01:
The getTimestamp() method does exactly that in the development branch, it interpolates a hardware timestamp based on the hardware rate and the high resolution software timer. 

I highly recommend developing for the development branch right now. There have been big changes on how channel objects and events work so any development using the master branch codebase will need big changes once the development branch goes into master.

shayo, 2017-08-01:
@aacuevas , last time I tried the development branch it was highly unstable and crashed a lot.
It doesn't make sense developing on top of that until that branch is stable. 
Adding on top of that is issues that still exist in the master branch which I would like to see solved before adding any additional functionality.

I am trying to keep most of my new code very self contained, so I hope once the dev branch is stable enough, it wouldn't be a lot of work to migrate to it.

It might be a good idea to keep a blog with all the changes that you are introducing to the development branch so things don't get implemented twice.

closed 2017-08-02 by shayo