open-ephys/plugin-GUI #157: OpenEphys_PluginGenerator fails to create project

rhayman, 2017-08-03:
I'm attempting to follow the tutorial instructions (https://open-ephys.atlassian.net/wiki/spaces/OEW/pages/72155138/Plugin+Generator) to create a new plugin with a build of the pluginGenerator. Plugin creation fails at the final step. Clicking the "Create project..." button results in nothing happening. I've tried various combinations with no luck.
I'm working with a copy of the plugin-GUI cloned about a week ago; Juce version reported as 4.2.1. My OS: Ubuntu 17.04

jsiegle, 2017-08-03:
@sept-en can you take a look at this?

jsiegle, 2017-08-08:
Any progress on this? I'll try to take a look at it tomorrow.

rhayman, 2017-08-08:
I re-compiled OE from scratch today following cloning the latest master branch but I've not yet had chance to see if I can re-create the issue...

Robin Hayman
Research Associate | Clinical & Experimental Epilepsy | ION

closed 2017-08-09 by rhayman