open-ephys/plugin-GUI #158: Crash when setting options on spike sorter

jerlich, 2017-08-16:
Just trying to play with the included data files, and the system crashes constantly.
Ubuntu Mate 16.04, built from source with 
"JUCE_DISABLE_NATIVE_FILECHOOSERS=1"
also using the libudev.so.1 hack for 16.04.

It's pretty frustrating. 

I think it would help a lot if there was a description of stable setups (e.g. hardware + OS + software versions).  

Steps to reproduce crash:
1) Add FileReader, Bandpass Filter, SpikeSorter
2) In SpikeSorter, add 4 tetrodes, set all thresholds at 100
3) Open SpikeSorter in tab
4) Play data
5) Drag threshold down -> SEGFAULT

jsiegle, 2017-08-16:
Which branch are you using, master or development? There are still a few unresolved issues with the SpikeSorter and the new channel objects in the dev branch. Either way, we'll get it fixed.

jsiegle, 2020-12-16:
Seems like this issue has been fixed.

closed 2020-12-16 by jsiegle