open-ephys/plugin-GUI #159: Crashes while playing or recording

lshaheen, 2017-08-24:
I've just started trying to get the Open Ephys system to work on a Windows computer. Everything works but then crashes sometime between 1 and 20 minutes into recording, giving the error: "CRITICAL: Timeout on pipe read. Check block and buffer sizes." I'm monitoring the system CPU and memory and they're not close to maxed out when this happens. The CPU is an i7-2700K and is running Windows 7 with 8 GB of RAM. The graph I'm using is pretty simple (see below). Any advice on how to get this working?

After it crashes I usually have to unplug and replug the power to the FPGA to get it working again. Is that normal?

[image]

jvoigts, 2017-08-24:
This problem could be either a software issue, a driver problem, or a hardware issue - i think this error message is also consistent with temporarily powering down/disconnecting the acquisition system, due to a faulty power supply etc. As a first test, could you check if you have the same issue either with a simpler pipeline, or with the intan software?

lshaheen, 2017-08-24:
Hi, thanks for the quick reply! I tried using close to the simplest possible graph (an FPGA and a channel map with the recording coming from the channel map) and it still crashed (see below). I can't get the Intan software to run (RHD2000interface v1.5.2), even after installing both MS Visual C++ 2010 and 2013 Redistributable I get the MSVCP120.dll is missing error.

[image]

aacuevas, 2017-08-24:
That error is common when usb bandwidth is not high enough.
Are you using a usb3 board? If so, is it plugged to an usb3 or usb2 port? Have you any other device plugged into the same internal usb hub (usually, contiguous ports in the motherboard share internal hubs)? 
The usb3 system has some optimizations to make good use of the high bandwidth with very high channel counts but we've heard of instances where those actually make usb3 boards struggle with mid-high channel counts like yours, perhaps this could be your case.

jvoigts, 2017-08-24:
Ah yes, good catch - you could also try just one headstage to narrow it down to a usb bandwidth problem.

lshaheen, 2017-08-26:
It's the usb3 board. It turned out that plugging it into a usb2 slot, and plugging the keyboard and mouse into the usb3 slots made it work. I still crashed when plugging the OEP board into a usb3 slot with nothing else in the other usb3 slots.

I'm already using one headstage - the Intan RHD2000. I'm actually only using 64 of the 128 channels but haven't found a way to turn off the other 64. I'm not recording them, but if they were not being communicated over the usb at all that would save some bandwidth I'd imagine. Is there a way to do this?

jsiegle, 2022-07-30:
This should be fixed now, but if you do experience this problem with the latest version of the GUI, please re-open this issue.

closed 2022-07-30 by jsiegle