open-ephys/plugin-GUI #161: File Reader Freezes Open-Ephys Linux Build

whitepine, 2017-09-24:
When trying to select a file to read from the "File Reader" plug-in, open-ephys crashes. I think this is because it by default uses a Window's convention for specifying the drive: "F:". Is there any easy fix for this?

aacuevas, 2017-10-02:
Does it crash when you click on the open file button? (The "F:" is not a Windows drive, but short for "File")
If so it might be due to the bug stated in #42 for which JUCEs dialog sometimes appear hidden, which makes the GUI appear to be frozen.
We added the compiler flag stated on that issue page to the build process but it seems that at some point got deleted without us noticing. Building the GUI again with that flag should solve your issue.

whitepine, 2017-10-02:
Your suggestion fixed the problem!

From: Aarón Cuevas López
Sent: Sunday, October 1, 2017 6:57 PM
To: open-ephys/plugin-GUI
Cc: whitepine; Author
Subject: Re: [open-ephys/plugin-GUI] File Reader Freezes Open-Ephys LinuxBuild (#161)

Does it crash when you click on the open file button? (The "F:" is not a Windows drive, but short for "File")
If so it might be due to the bug stated in #42 for which JUCEs dialog sometimes appear hidden, which makes the GUI appear to be frozen.
We added the compiler flag stated on that issue page to the build process but it seems that at some point got deleted without us noticing. Building the GUI again with that flag should solve your issue.

closed 2018-02-20 by whitepine