open-ephys/plugin-GUI #162: Crash when using serial port

rwd122, 2017-09-24:
When I try to use serial port as a sink and LFP Viewer as a sink, the program crashes after 1-2 seconds. The incoming signal is from an arduino. Windows 10.

FlorianFranzen, 2017-09-28:
As far as I know, serial port data can not be used with the LFP viewer. What are you trying to achieve?

rwd122, 2017-09-28:
Thanks for the reply. I am trying to send in an analog signal from arduino to GUI via the serial port. The serial communications are working properly. GUI crashes anytime I use serial port as the source (even when I don’t use LFP Viewer). It crashes on multiple computers. Should this be able to work?

Ryan

aacuevas, 2017-10-02:
The serial node, as it works now, does not actually generate continuous signals. but each time a message is received it creates a binary type event with the receiver buffer. It was originally created to send extra data from some hardware to the GUI alongside continuous data acquired from somewhere else, using a merger, but not to generate actual continuous channels from it. 

The current master version of the GUI has a couple of known bugs regarding signal chains with no continuous channels. This has been fixed in the development version, but you still won't be able to stream data from the serial port. It should be feasible to create a simple source node processor that buffers from the serial port, though.

AGenews, 2018-09-20:
Hi,
did anything change regarding a serial data stream plugin? Are there any plans to implement such a thing? It would be nice to simply hook up a bunch of I2C sensors via an Arduino and send the values to the OE GUI along the neural recordings.  The way the "Serial Plotter" in the current Arduino IDE (1.8.5) works is pretty amazing.
Because it kinda feels terrible to use a DAC to transfer the data from the Arduino to  OE again ;)

Best and many thanks!
Andreas

jsiegle, 2022-07-30:
The Serial Port plugin has been deprecated as of the latest version of the GUI. If you need this plugin for your experiments, please open a new issue.

closed 2022-07-30 by jsiegle