open-ephys/plugin-GUI #171: Testing 0.4.3 : no debug64 version?

vncntprvst, 2017-11-06:
Hi everyone,
I tried compiling the new version (0.4.3) on Windows. The Debug works fine (although there are issues with plugin compilation later on), but Debug64 fails, saying that it can't find the path specified. 
[image]
Is there no debug64 version? 
Thanks
Vincent

vncntprvst, 2017-11-06:
Related to this issue: when compiling plugins, x64 version are not available, and fail as expected:
[image]
However, even setting the plugin project to Debug / Win32 in Visual Studio's Configuration manager, there are errors: 
[image]

aacuevas, 2017-11-06:
Please, check that the architecture selector (just at the right of the Release/Debug selector) is set to x64 when compiling the Debug64 version.

Also, you won't be able to start the GUI from the plugins project, that's normal.

vncntprvst, 2017-11-06:
OK, thanks. My bad. There's no architecture selector in the menu bar (at least by default), it's in the Configuration Manager and I had forgotten to check that - I don't compile stuff very often. The GUI works now. 
[image]. 
Only thing is, it crashes when I try to monitor the audio signal. I might have missed something in the compilation.

closed 2017-11-10 by vncntprvst