open-ephys/plugin-GUI #173: Merger utility causes crashes when reordering signal chain

kmichaelfox, 2017-11-06:
Re-configuring signal chains with Mergers in them is currently easy to break, causing seg faults.

The two most observable behaviors right now are:
1) Placing a Source or Filter processor before a Merger then deleting the Source/Filter causes a crash. (Deleting 1 of 2 branch inputs works, but deleting the last Merger input breaks.)
2) Mergers with 1 input source will not crash if they are dragged to the left of their input, but will still cause a crash when the Merger is subsequently deleted.

Neither of these behaviors occur if the Merger is deleted without ever receiving an input processor.

The mergers also exhibit unclear logic in how branches are displayed in the signal chain channels (A, B, C, D). Orphans nodes can live on invisible sub-chain channels in the signal chain drawer, but which can only be selected via the Node Graph canvas. It is not clearly if this issue is or is not related to the crashing behaviors described above.

aacuevas, 2017-11-07:
I think I've fixed this (in the testing branch). Can you please check for all your test cases that it is indeed fixed?

kmichaelfox, 2017-11-07:
It passes my test cases now.

closed 2017-11-07 by kmichaelfox