open-ephys/plugin-GUI #175: Spike Sorter causing crashes when attempting to monitor with incomplete config/inputs

kmichaelfox, 2017-11-06:
Spike Sorter consistently crashes if monitor is clicked and it has neither: An appropriate Source processor input; nor, a user configured electrode.

Additionally, Spike Sorter crashes the GUI application with or without monitoring enabled, when:
1) a valid Source processor input is provided; then,
2) a valid electrode configuration is applied; then,
(optionally) monitor is enabled; then,
3) the Source input is deleted

aacuevas, 2017-11-07:
The crashes should be fixed now in the testing branch. Please check if they persist in any of your test cases.

kmichaelfox, 2017-11-07:
The crashes are taken care of for my tests. Although, deleting the Source input allows the Spike Sorter to remain in the previous signal chain's configuration. For instance, allowing monitor to remain ON and retaining electrode graphs.

Since Spike Sorter disallows users to create electrodes or turn on monitoring when it does not have an input, is this the desired behavior? Or should it revert completely to its default construction state for safety, including disabling monitoring?

aacuevas, 2017-11-07:
It is desired behavior, in case users want to update the sources without having to recreate all the electrode configurations. The sorter has a safety that prevents acquisition in case there were not enough input channels for all existing electrodes.

kmichaelfox, 2017-11-07:
Perfect.

closed 2017-11-07 by kmichaelfox