open-ephys/plugin-GUI #176: Processors drawer exhibits resizing problems when window width is sufficiently small

kmichaelfox, 2017-11-06:
The contents of the Processors drawer spills out of it's component container when the window is close to its minimum width.

For most window sizes, resizing the width causes the canvas area to stretch with minimal impact to either the Signal Chain or Processors drawer. After decreasing the width beyond a certain point, the canvas area attempts to maintain a fixed width and begins to displace the Processors drawer off the screen to the left if it is open. After this threshold, the Processors drawer overrides its height constraints as well, causing it to drawn outside of the intended bounds.

This behavior is exhibited both with the Signal Chain drawer open or closed.

Before reaching width threshold:
[image]

After passing width threshold, Signal Chain drawer open:
[image]

After passing width threshold, Signal Chain drawer closed:
[image]

aacuevas, 2017-11-07:
Well, resizing beyond a certain point did cause other issues too, so I've increased the minimum window size. I'm not getting this or other issues now. Of course, this does not apply to Linux, due to #177

kmichaelfox, 2017-11-07:
Clamping the window's width to a minimum of 484 solved this problem for me on all three platforms. #179 solves this issue for linux on Debian 9.2.1.

closed 2017-11-29 by kmichaelfox