open-ephys/plugin-GUI #178: testing v0.4.3 - Spike Sorter crashes GUI when monitoring audio

vncntprvst, 2017-11-07:
I failed to mention in the last post of issue #171 that the crash is induced from monitoring audio in Spike Sorter. This similar to issue #175, except all inputs are correct (as tested in latest binary release).

aacuevas, 2017-11-07:
Audio monitoring in general is known to be unstable on some systems, specially when sending more than one channel to the audio system (which is the case when you enable audio monitoring on a multichannel electrode on the sorter). A complete rework of the software audio output architecture is on our TODO list.

Does it also crash for you if you try to monitor multiple channels from any other processor?

vncntprvst, 2017-11-07:
Yes, unfortunately, it does crash when selecting the Spike Detector's Monitor option, as well as when selecting a channel in the Audio thumbnail display of other processors. 
I also tried monitoring a single channel in Spike Sorter, with the same outcome.

aacuevas, 2017-11-07:
I just checked and there is a new bug introduced in the testing branch that causes the GUI to crash when selecting a channel for audio monitoring while the GUI is acquiring. That didn't happen before. We'll fix it before release.

vncntprvst, 2017-11-07:
OK. Thanks. Just for the record, the audio test in the audio parameters window works fine.

aacuevas, 2017-11-07:
The crash should be fixed now in the testing branch, but don't expect the audio to actually sound like it should. As I said, the audio output system is in dire need of a complete rework.

vncntprvst, 2017-11-09:
OK, thanks. I'll test it.