open-ephys/plugin-GUI #186: Crash when trying to initialize Pulse Pal output when device not plugged in or driver not installed

ethanbb, 2017-11-20:
There's a crash when trying to initialize the Pulse Pal output plugin when the device isn't plugged in and ready to use. This is due to lack of error checking for whether there any COM/LPT devices available, in this code (PulsePal.cpp, `PulsePal::initialize()`):

```c++
    vector<ofSerialDeviceInfo> devices = serial.getDeviceList();
   // bool foundDevice = false;
    int id = devices[0].getDeviceID();
```
If `devices` is empty, there's an assertion failure. Instead of letting this happen, we should make an error message pop up in the GUI, like when you try to add a Rhythm FPGA source and there's no acquisition board connected.

alejoe91, 2018-01-17:
Hi, is it fixed with the new Pulse Pal Output implementation (b291e8fb6c5aefe2a8d6fa5922d8a06b018c8721)? In the new implementation I check all devices and if the Pulse Pal is not found it prints **NOT CONNECTED** in the visualizer.

https://github.com/open-ephys/plugin-GUI/blob/development/Source/Plugins/PulsePalOutput/serial/PulsePal.cpp#L97-L131

ethanbb, 2018-01-24:
Yup, it seems to be working now. Thanks!

closed 2018-01-24 by ethanbb