open-ephys/plugin-GUI #190: Fail Audio (GUI-Testing Branch)

410pfeliciano, 2017-12-22:
Adding audio callback.
Audio node setting channel to 201
JUCE Assertion failure in juce_AudioSampleBuffer.h:638
Segmentation fault (core dumped)

This only happens in the Testing and Developing branch of the plugin-GUI when I try to listen (software) a channel. The GUI is installed in Ubuntu 14.04LTS.  No audio problems with the master GUI.

kmichaelfox, 2017-12-25:
When you say there are no audio problems with the Master GUI, are you referring to the old Master branch or the most recent one? The most recent Master branch includes updates from Testing and Development.

Also, which processor are you attempting to monitor the audio stream from?

jonnew, 2018-02-19:
I'm getting this as well. It happens even if there are no processors. I'm using a custom multi-subprocessor source.

jsiegle, 2018-02-19:
I think I fixed it. In line 292 of AudioNode.cpp, the destination channel should have been 0, rather than i, because the temporary buffer only has one channel. This was incorrect on the master, testing, and development branches.

Audio monitoring works for me now. Can someone else confirm that this fixes the problem?

aacuevas, 2018-02-19:
Yup, I was looking at the same thing and I think it solves the issue. If everything is correct I'll probably update the binaries with this tomorrow.

jonnew, 2018-02-19:
Yes, I can confirm this fixes it.

410pfeliciano, 2018-02-19:
Yes, the issue has been fixed.

closed 2018-02-19 by 410pfeliciano