open-ephys/plugin-GUI #195: Support for Intan RHS2000 Stimulation / Recording Controller

beOn, 2018-01-08:
This has come up before, but it's been a while and it's not in the issue tracker, so here it is.

Some links:
* [Previous Conversation from Dec 2016](https://groups.google.com/forum/#!topic/open-ephys/DaoE02V45K4)
* [Intan Stimulation / Recording Controller Page](http://intantech.com/stim_record_controller.html)
* [Intan RhythmStim USB/FPGA Interface](http://intantech.com/products_RHS2000_RhythmStim.html)

Before we go nuts here - has there been any follow-up to the conversation from late 2016? Presumably just adding support for the controller would still be pretty easy. But has anyone worked on the main challenge, pointed out by @jvoigts: stimulus design and delivery.

angelonc, 2018-11-29:
Hi, I'm also interested if there has been any follow up on this?

Also, is the current architecture for the RHD2000 compatible with the RHS2000? That is, can I just record using the RHS stimulation system with the current build of the plugin-GUI?

hotsoupisgood, 2019-12-15:
Any progress on this?

jsiegle, 2022-07-30:
We are currently working on this in the [`refactoring`](https://github.com/open-ephys-plugins/rhythm-plugins/tree/refactoring) branch of the Rhythm Plugins repository. We will send out an update when support for the Stim/Record controller has been added.

closed 2022-07-30 by jsiegle