open-ephys/plugin-GUI #196: Error finding FPGA: "okUsbFrontPanel: Could not claim interface"

byronprice, 2018-01-30:
I recently merged my branch with the upstream branch and now get the following error:

Loading processor graph.
Clearing signal chain.
Item dropped at insertion point 0
Creating from plugin info...
(-1)::Rhythm FPGA
/home/jglab/plugin-GUI/Builds/Linux/build
---- Intan Technologies ---- Rhythm RHD2000 Controller v1.41 ----

FrontPanel DLL loaded.  Built: Mar 17 2014  12:03:57

Scanning USB for Opal Kelly devices...

Found 1 Opal Kelly device connected:
  Device #1: Opal Kelly XEM6310LX45 with serial number 1738000IHN

FPGA system clock: 0 MHz
Opal Kelly device firmware version: 1.30
Opal Kelly device serial number: 1738000IHN
Opal Kelly device ID string: Opal Kelly XEM6310

Rhythm configuration file successfully loaded.  Rhythm version number: 1

Initializing acquisition board.
Blocksize set to 1024
DDR burst set to 32
Sample rate set to 30000
USB3 board mode enabled
Sample rate set to 30000
Number of enabled data streams: 8
Checking for connected amplifier chips...
Number of enabled data streams: 1
Sample rate set to 30000
  Adding node to graph with ID number 109

Rhythm FPGA updating settings.
Rhythm FPGA setting num outputs to 35
Setting buffer address to 0x1ffacf0
Input buffer address is 0x1ffacf0
Rhythm FPGA updating settings.
Rhythm FPGA setting num outputs to 35
Item dropped at insertion point 1
Creating from plugin info...
(-1)::Rhythm FPGA
/home/jglab/plugin-GUI/Builds/Linux/build
---- Intan Technologies ---- Rhythm RHD2000 Controller v1.41 ----

FrontPanel DLL loaded.  Built: Mar 17 2014  12:03:57

Scanning USB for Opal Kelly devices...

Found 1 Opal Kelly device connected:
  Device #1: Opal Kelly XEM6310LX45 with serial number 1738000IHN

**okUsbFrontPanel: Could not claim interface
Device could not be opened.  Is one connected?**

aacuevas, 2018-01-30:
It seems as if it's trying to open the acquisition board twice. Does the issue occur when dragging the Rhythm node to the signal chain or when loading a prior signal chain?

byronprice, 2018-01-30:
I've got it set to automatically load a signal chain. I can't get it to open without doing so. Then, it gets stuck with the "Cannot find acquisition board" window.

aacuevas, 2018-01-30:
You can always delete or move the lastConfig.xml file, which contains the last loaded signal chain. 
Actually, please move it and send it to me ( to aaron@open-ephys.org ) so I can see what is causing the issue. I've tried making a chain with an Rhythm node and autoreloading it myself and I cannot seem to reproduce this.

closed 2022-07-30 by jsiegle