open-ephys/plugin-GUI #197: Generaly instability, audio artifacts, etc

jonnew, 2018-02-20:
Revision:  8406ede366ef025c8754f17e56eb52510739524e
OS: Ubuntu 16.04

**Many breaking changes** have recently been introduced into the OE code base. I've come back to OE GUI after a pause and updated to latest master. The stability of OE GUI has never been great. However, recent changes have made it basically unusable. Some things I've noticed anecdotally. I will add more if I encounter them.

**Audio artifacts:** 
- Sometimes, when I select a channel audio sounds like a low hum with pops on it. I restart the GUI it it sounds more normal.
- No matter what, when I start, audio sounds normal then fades into a jumpy sporadic sounding version of what should be there over about 1-2 seconds. If I restart, the process repeats.

**LFP Viewer Broken**
- For multi-subprocessor sources, making a selection on the LFP visualizer does nothing.

**Segfaults everywhere, issues with processing graph**
- This has always been true but feels even worse now
- Adding/removing upstream elements in signal chain have about a 50% chance of resuling in segfault when chain is run after the modification
- It is possible to get floating graph sections when certain elements are deleted and its impossible to reconnect them as far as I can tell:
[image]

**Digital referencing is broken**
- For certain channel maps (especially those with removed channels), I cannot select a digital reference. It seems like it is trying to reference a channel that is not there or something
- For multi-subprocessor sources with the same sample rate, digital referencing does not take phase of buffers into account and spurious subtractions result (i.g. there is no equivalent to a `zip` operation occurring before subtraction)

jsiegle, 2018-02-20:
Thanks for putting this list together. My guess is that most of these issues are only seen when using a source with multiple subprocessors, a configuration that hasn't been extensively tested yet. The Neuropixels module has 2 subprocessors (spike band and LFP band), but I've only tested it with simple signal chains.

@aacuevas, I think we need to create a multi-subprocessor File Reader ASAP that can be used to troubleshoot these issues.

I have a version of the LFP Viewer that is compatible with multiple subprocessors, I will push that to GitHub later today.

@jonnew, what are the details of the subprocessors in your source node (e.g., # of channels per subprocessor, and the sampling rate of each)?

jonnew, 2018-02-20:
For the digital referencing issue I agree and it seems that the reason is pretty obvious. For the rest, I'm not so sure. I can see in the commit log that there have been a number of "improvements" made recently to the core facilities of the GUI. I would start there.

jonnew, 2018-02-20:
With respect to source: 2 subprocessors. 27kHz each.

jonnew, 2018-02-20:
**Selecting "none" does not change audio**
- In Audio part of processor drawer, if non is selected the channel is de-highlighted but audio continues to be produced. Restarting processing chain causes audio to stop.

jsiegle, 2018-03-12:
Just an update on where things stand with this...

We think most of these issues are being caused by a mismatched number of samples between subprocessors. Rather than forcing every plugin to check for this, we are going to update the splitter/merger architecture to deal with subprocessors more sensibly.

The first change will be to allow splitters to split the output of multi-subprocessor plugins into separate signal chains. This will prevent processors from performing buffer operations across channels with different numbers of samples. In your case, this would disallow channels from each subprocessor from being referenced to one another.

The second update will be to give mergers the ability to match the number of samples per buffer across subprocessors or processors. If two data sources with the same sample rate have a mismatched number of samples in a particular buffer, any extra samples will be saved for the start of the next buffer. If two data sources with different rates are merged, data points will be added to the buffer so the number of samples for each is the same. This will allow any combination of channels to be referenced to one another, but comes at the cost of a slight delay as well as the need to add artificial data points when the sample rates do not match.

These updates should solve all of the bugs except for those related to audio monitoring. For that, we'll have to switch the order of operations so that up-sampling to the rate of the audio card happens prior to adding channels together. Then, monitoring can happen across processors with arbitrary combinations of sample rates.

@aacuevas should have an estimate of how long this will take in the next few days. The splitter update is the highest priority, and the merger update will happen later.

jonnew, 2018-03-12:
cool. I think maybe the JUCE data processing architecture is just kinda
limited for this type of thing. I actually think that just forcing sources
that expose multiple processors (i.e. feed multiple buffers) to occupy
parallel processing streams (no merging) might be the right way to go with
JUCE. IMO it is extremely clumsy for dealing with asynchronous streams and
requires a great deal of manual tricky coding to get right rather than it
being implicit in the core functionality of the library.

This also does not account for all the segfaults, but i think these are
probably a result of the weakness of the OE interfaces for creating
plugins. They provide very little control (or even guidance) over the
operations  the user may perform when overriding methods. I dunno though,
just guessing.

ethanbb, 2018-07-20:
@jsiegle This is old and I don't know how far you've gotten in implementing new behavior for mergers, but: adding artificial data points to fix sample rate mismatches seems problematic and rarely useful. Instead I would suggest not allowing merging streams with different sample rates, but adding downsampler and upsampler plugins, so that the sample-rate-matching is explicit and doesn't add gaps to the data.

As for merging streams with same sample rate but potentially different numbers of samples for each buffer, saving extra samples makes sense to me. In the general case then, the merger would store a queue of "extra data" for each channel being merged, and on a given callback, only the max number of samples that all channels have available would be passed through the merger for further processing.

jsiegle, 2022-07-31:
The problems detailed in this issue were a major motivation for the switch to "stream" based processing in GUI version 0.6.0. Previously, there was limited support for asynchronous data sources using "subprocessors," but we've now made the [`DataStream`](https://github.com/open-ephys/plugin-GUI/blob/main/Source/Processors/Settings/DataStream.h) class a central part of the GUI's functionality.

Every continuous, event, and spike channel *must* be associated with a `DataStream`, which represents a block of continuous channels that are guaranteed to have the same number of samples in every buffer. If multiple sources are merged together, channels from each source will be associated with different streams, and therefore will be handled separately by downstream plugins.

It's now much easier to ensure that operations such as referencing or spike detection are only performed across synchronously sampled channels. Instead of looping through all channels in the `process()` method, plugins should first loop through all incoming `DataStreams`, and then loop through each channel within a given stream. Furthermore, it's simple to apply parameter values on a per-stream basis, e.g. to have different filter settings for different streams.

See the [Common Average Reference source code](https://github.com/open-ephys/plugin-GUI/blob/5706680e8dbacafde0bb43a2839c5ae897d0646b/Plugins/CommonAverageRef/CommonAverageRef.cpp#L74-L134) for an example of how this should be implemented.

closed 2022-07-31 by jsiegle