open-ephys/plugin-GUI #198: Increasing software buffer size results in seg-fault

whitepine, 2018-02-20:
The default software buffer size is 1024 samples. Users can change the software buffer size by clicking on the latency (default is "23ms"). When I try to increase the buffer size to say 2048 samples, open-ephys seg faults. My data source is "file reader". Interestingly, OE doesn't crash when I decrease the buffer size. 

I thought that the recent commit titled "Fix destination channel of copying to temp buffer" might have fixed the issue. I implemented the change from this commit to my forked version of OE and it still crashes.

Also, I tried increasing the default buffer size to 1024 by changing line 185 in Audio Node.cpp

filters.add(new Dsp::SmoothedFilterDesign<Dsp::RBJ::Design::LowPass, 1> (1024***2**))

but it does not appear to have had an affect on the buffer size.

jsiegle, 2018-02-22:
Are you using the File Reader, by chance? For that source node, it seems like the problem is resulting in the hard-coded `BUFFER_SIZE` constant on line 34 of FileReader.h. If I increase that to 2048, I can bump the audio buffer size up to 2048 without any problems. Higher than that, and it crashes again. Obviously that's just a temporary fix...we need to figure out a way for the FileReader to update its buffer size depending on the audio buffer.

Also, the buffer size is set in the Audio Component (which interfaces with audio hardware), not the Audio Node (which outputs data for audio monitoring). You can change it in line 85 of AudioComponent.cpp, but that didn't fix the problem.

whitepine, 2018-02-22:
Ahhh I see. Thanks a lot for the help!

ethanbb, 2018-10-08:
Is this fixed now? There were some changes to fix issue #217 that I think got rid of that 1024 constant.

jsiegle, 2020-12-16:
This bug is still present, if you change the buffer size while acquisition is active. We need to disable the settings interface during acquisition.

anjaldoshi, 2020-12-18:
Original bug was already fixed by Aarón in commit  5a07d90, but as mentioned by Josh mentioned, changing buffer size while acquisition is active results in a crash. Fixed in #396.

commit ba184b69 (2020-12-23): Disable AudioEditor when acquisition is active. Closes #198
closed 2020-12-23 by anjaldoshi