open-ephys/plugin-GUI #203: TTL Event Offset Issues

angelonc, 2018-04-05:
I'm trying to synchronize some neural signals with stimulus events, and I'm using several channels of the TTL input to do so. However, I'm getting some instability when aligning the timestamps of the events from the spike times.

A few details first:
-Windows 10
-precompiled GUI (downloaded late 2017)
-flat binary format recordings

So, there have been a few posts about how TTL event times are recorded relative to the start of the acquisition (play button), while spikes are recorded relative to the recording onset. In one of my recordings, subtracting the recording offset from my event times (I read the sample of the recording start from 'sync_messages.txt', I'm recording in flat binary form) gives me nice event synchronization. However, another recording I made recently, accounting for the offset in 'sync_messages.txt' desynchronizes everything, but everything is fine when I don't remove that offset. 

I'm a bit puzzled as to why this is happening. I started both recordings the same way, starting an acquisition, plugging in my microdrive, and then pressing record after a little while. I did notice in the 'sync_messages.txt' of my recording that isn't synced if I account for the record onset, that the recording onset sample was exactly 2200000, which seemed a bit fishy to me, but I guess coincidences happen.

Has anybody else encountered this issue?

aacuevas, 2018-04-05:
I am currently investigating an issue in which sync_messages.txt start times might be inaccurate for the binary format in some cases.

Regardless of that, both spike and TTL times should not need any offset removal in this format in particular, since their timestamps are copied directly from the hardware-generated sample timestamps.

angelonc, 2018-04-06:
Thanks for your fast reply!

Sorry, I forgot to mention that the spike times I'm using are from KiloSort, which are referenced relative to the start of the recording. You're saying that the spike times generated from the Spike Detector in openEphys GUI are relative to the acquisition onset?

jsiegle, 2022-07-31:
This should no longer be happening, but if you do experience this with the latest version of the GUI (0.6.X), please re-open this issue.

closed 2022-07-31 by jsiegle