open-ephys/plugin-GUI #205: Channel names create conflicts in Open Ephys format when recording after a merger

ethanbb, 2018-04-05:
Steps to duplicate: Put two sources in a signal chain, join them with a merger, add a processor after the merger and record from this in Open Ephys format.

Result: If channels with the same original index from both source processors are recorded, their continuous filenames collide and the resulting file ends up corrupted.

Instead of using the original channel names, the `OriginalRecording` record engine should use the "real channel" indices, i.e. the indices of the channels as they are ordered in the processor being recorded from. However, this would eliminate the "ADC" and "AUX" labels for special channels. Maybe there needs to be a renaming system to deal with channels being merged more robustly. I know some details of the new channel system are still being worked out, so this is something to be aware of for that discussion.

aacuevas, 2018-04-05:
Actually, once the binary format is considered fully stable and has support scripts available on the tool repository, we're planning on considering the open ephys format a legacy one, making binary the default format.

Although we will try to solve issues like this, the open ephys format predates many of the current capabilities of the GUI, making it quite difficult to integrate them within the format. For this issue in particular, we will probably add some extra field to the filename to differentiate source node and recorded node, that should prevent conflicts.

ethanbb, 2018-04-06:
Makes sense. Of course I've noticed that the Open Ephys format has a lot of quirks, although I hadn't really tried testing edge cases until recently. We've been sticking with it for now thought because it's the easiest format to get into MATLAB with the current analysis tools. I've been trying to modify the script to convert Open Ephys files to KWIK so that it also converts AUX and ADC channels, which is how I ran into this.

So is the goal basically to have the new binary format replace Open Ephys format and NWB replace KWIK as the HDF5 option, once they're ready?

jsiegle, 2022-07-30:
This is fixed in the latest version of the GUI -- the Open Ephys format now supports multiple streams being merged together, as well as multiple streams generated by the same source (e.g. Neuropixels plugin).

closed 2022-07-30 by jsiegle