open-ephys/plugin-GUI #206: Missing text events

angelonc, 2018-04-11:
Specs first:
-Windows 10
-compiled openEphys GUI from late 2017
-flat binary recording format

I'm getting some instability with the GUI recording manually entered text events. During some recordings, a folder under events>recording1>Message_Center-904.0>TEXT_group**1** will be written, which has the timestamps and text strings of manually entered recording events.

However, in some other recordings, the same folder will be made, but will be called events>recording1>Message_Center-904.0>TEXT_group**2**, which will have empty text and timestamp files. While not a fatal issue for me, it is very useful when segmenting my recordings to have these events

aacuevas, 2018-04-13:
Thanks for the report. It was caused by some internal indexing in the message center not being properly cleaned. I've fixed it in the development branch, please check if it works for you now.

angelonc, 2018-10-01:
@aacuevas, thanks for looking into it. I compiled the most recent development version and used it for a bit without issue, but started recording with some large probes and the messages are still not being saved, even after creating fresh probe configs.

jsiegle, 2022-07-30:
If you experience this problem in the latest version of the GUI, please re-open this issue.

closed 2022-07-30 by jsiegle