open-ephys/plugin-GUI #218: FileReader does not react to buffer size when filling first buffer cache

ethanbb, 2018-05-10:
Related to #217. The FileReader always assumes 1024 (`BUFFER_SIZE`) samples per buffer when filling the first buffer cache on loading a file. The lines in question (in FileReader::setFile):

```
    m_samplesPerBuffer.set(float(BUFFER_SIZE) * (getDefaultSampleRate() / 44100.0f));
    
    readAndFillBufferCache(bufferA); // pre-fill the front buffer with a blocking read
```

The result is that, if the buffer size is not actually 1024, either not enough or too many samples are loaded into `bufferA`, causing either skipped data or a gap in data. Instead of using a constant `BUFFER_SIZE` and sample rate of 44100 (see #217), the FileReader should read both values from the AudioDeviceManager and update if they are changed. To make it simpler, the first buffer could be filled with more samples than needed initially, and then just `samplesRead` could be updated if the audio settings changed.

(Even though this may not be such a serious bug, I feel it's better to document what's going on in case it saves someone some time in the future.)

aacuevas, 2018-05-15:
Fixed with #217 Bufers should have the correct values now

ethanbb, 2018-05-15:
Awesome, thanks for addressing these FileReader issues! I'll verify that the three I submitted are fixed.

closed 2018-05-15 by ethanbb