open-ephys/plugin-GUI #219: RHD2164 headstage not detected in latest dev

mspacek, 2018-05-11:
I just grabbed and built (after a `make clean`) the latest dev version. When I run it, our USB3 open-ephys board is detected, but the RHD2164 headstage connected to it (Port A) isn't detected when I add the Rhythm FPGA to the signal chain. Hitting rescan doesn't help. It works fine on an older version of the plugin-GUI from the dev branch from about March 2017. Has something major changed since then? I see some of the `.bit` files have been renamed...

This is in Xubuntu 16.04. We have the Open-Ephys acquisition board 2.2 rev 2.

Here's the output of gdb when I run it and add the Rhythm FPGA source to the signal chain:
```
(gdb) run
Starting program: /home/mspacek/src/oe.master/Builds/Linux/build/open-ephys 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
JUCE v4.2.1

Created processor graph.

[New Thread 0x7fffec7db700 (LWP 10437)]
[Thread 0x7fffec7db700 (LWP 10437) exited]
[New Thread 0x7fffec7db700 (LWP 10438)]
[New Thread 0x7fffebfda700 (LWP 10439)]
Got audio device.

Audio device name: Playback/recording through the PulseAudio sound server
[Thread 0x7fffebfda700 (LWP 10439) exited]
[Thread 0x7fffec7db700 (LWP 10438) exited]
[New Thread 0x7fffec7db700 (LWP 10440)]
[Thread 0x7fffec7db700 (LWP 10440) exited]
[New Thread 0x7fffec7db700 (LWP 10441)]
[New Thread 0x7fffebfda700 (LWP 10442)]
Audio device type: ALSA
Audio output channels: 2
Audio device sample rate: 44100
Audio device buffer size: 1024

Created audio component.
Created message center.
Created info label.
Created graph viewer.
Adding tab with index 1
Adding tab with index 2
Created data viewport.
Created filter viewport.
Created control panel.
Created filter list.
Created plugin manager
Loading Plugin: PhaseDetector... Loaded with 1 plugins
Loading Plugin: RecordControl... Loaded with 1 plugins
Loading Plugin: ArduinoOutput... Loaded with 1 plugins
Loading Plugin: SpikeSorter... Loaded with 1 plugins
Loading Plugin: IntanRecordingController... Loaded with 1 plugins
Loading Plugin: BinaryWriter... Loaded with 1 plugins
Loading Plugin: PulsePalOutput... Loaded with 1 plugins
Loading Plugin: KWIKFormat... Loaded with 2 plugins
Loading Plugin: FilterNode... Loaded with 1 plugins
Loading Plugin: NWBFormat... Loaded with 1 plugins
Loading Plugin: CAR... Loaded with 1 plugins
Loading Plugin: ChannelMappingNode... Loaded with 1 plugins
Loading Plugin: Rectifier... Loaded with 1 plugins
Loading Plugin: NetworkEvents... Loaded with 1 plugins
Loading Plugin: EventBroadcaster... Loaded with 1 plugins
Loading Plugin: RhythmNode... Loaded with 1 plugins
Loading Plugin: LfpDisplayNode... Loaded with 1 plugins
Loading Plugin: BasicSpikeDisplay... Loaded with 2 plugins
Loading Plugin: EvntTrigAvg... Loaded with 1 plugins
Loading Plugin: SerialInput... Loaded with 1 plugins

Loading window bounds.

[New Thread 0x7fffdf4f1700 (LWP 10444)]
Item dropped at insertion point 0
Creating from description...
Sources::Rhythm FPGA (2-1)
Rhd2000EvalBoardUsb3: Allocating 33.153 MBytes for USB buffer.
/home/mspacek/src/oe.master/Builds/Linux/build
---- Intan Technologies ---- Rhythm RHD2000 USB3 Controller v2.0 ----

FrontPanel DLL loaded.  Built: May 25 2017  14:06:51

Scanning USB for Opal Kelly devices...

Found 1 Opal Kelly device connected:
  Device #1: Opal Kelly XEM6310LX45 with serial number 1646000GEU

Attempting to connect to device '1646000GEU'
Opal Kelly device firmware version: 1.26
Opal Kelly device serial number: 1646000GEU
Opal Kelly device ID string: Opal Kelly XEM6310

Rhythm USB3 configuration file successfully loaded.

Initializing acquisition board.
Sample rate set to 30000
USB3 board mode enabled
Scanning
Sample rate set to 30000
Number of enabled data streams: 16
Checking for connected amplifier chips...
Number of enabled data streams: 0
Sample rate set to 30000
  Adding node to graph with ID number 100

Rhythm FPGA updating settings.
Rhythm FPGA setting num outputs to 0

```

aacuevas, 2018-06-01:
I've discovered an issue where, in linux, sometimes the Intan Recording Controller and the RhythmNode plugins can interfere, even if they aren't being used at the same time. It might be related.
While I fix the issue, clean both the base GUI (make clean) and the plugin build files (make -f Makefile.plugins clean), disable the Intan Recording Controller plugin by removing or renaming the Makefile in Source/Plugins/IntanRecordingController and build everything again from scratch. See if this solves your issue.

mspacek, 2018-06-05:
Hi Aaron, sorry for the delay. I finally gave it a try, and it works! Thanks!

closed 2020-12-16 by jsiegle