open-ephys/plugin-GUI #224: Channel labels don't propagate between blocks

mspacek, 2018-06-05:
I only recently realized this after messing around with the channel map plugin and noticing it wasn't changing the labels in the LFP viewer. From the channel block's [help page](https://open-ephys.atlassian.net/wiki/spaces/OEW/pages/950421/Channel+Map)

> To rearrange channels, click REMAP (it will become darker). You can then re-order channels by dragging. In subsequent blocks, "channel 1" will refer to the first channel in the list (in the example picture above, channel 2).

IMO, this is a fatal flaw, and makes open-ephys nearly unusable. How is anyone supposed to know what they're looking at in e.g. the LFP Viewer if the "channel numbers" are always labelled 1..nchans, regardless of how they were rearrange by the preceding channel map block? After staring at the code, I'm not sure the infrastructure is even there to propagate actual channel numbers (i.e. labels) between blocks, instead of always recalculating channel indices for each block. Is this feasible?

jsiegle, 2018-06-05:
I think the propagation of channel names was lost during our recent overhaul of the LFP Viewer. It should be possible to add it back...all of the processors have access to this info through the "channels" array. I'll take a look at it in the next few days.

mspacek, 2018-06-06:
Good to hear! I just tried an older version, something like 0.4.1 from about January 2017, and the LFP Viewer (not the LFP Viewer Beta) block seems to have the same behaviour: reordering or disabling channels doesn't change channel labels in the LFP Viewer. Disabling only changes the total channel count.

I noticed that `DataChannel->getName()` returns 1-based labels like `CH1`, `CH2`, etc, and that these seem to be preserved with changes in channel order and enable state. I've made some modifications to the LFP Viewer to use the chan name instead of the chan number...

jsiegle, 2020-12-16:
This is fixed now.

PR #225 (2019-09-05): Label LFP Viewer chans by name instead of number
My attempt to fix #224. Sorry about the whitespace, the mix of tabs and spaces drives me nuts in these two files, so I replaced all tabs with 4 spaces. This seems to more or less be the standard in the project.
closed 2020-12-16 by jsiegle