open-ephys/plugin-GUI #226: Put back beta lfp viewer

jonnew, 2018-06-08:
Current lfp viewer is broken for multi processor sources.

jsiegle, 2018-06-11:
I'm fine with adding it back to the dev branch for now, but we will also need to figure out what's wrong with the new viewer. What configuration breaks it?

jonnew, 2018-06-11:
I don't agree. I think that breaking changes should probably not take precedence over stable but limited code. 

The major issue is that I cannot change the sub-processor that I'm looking at. Changing it in the drop down menu has no affect on the channels that are shown.

jsiegle, 2018-06-12:
That functionality is working in the dev branch... @aacuevas, do you think we could be ready for another master release in the next week or so?

In the meantime, I added the beta LFP Viewer back to the dev branch, because it has the ability to display multiple subprocessors simultaneously, which the newer LFP Viewer does not have.

closed 2020-12-16 by jsiegle