open-ephys/plugin-GUI #232: BinaryWriter not detecting spikes on channels other than 0 and 2

mspacek, 2018-06-28:
While testing some of my recent changes against the dev branch, I noticed that the BinaryWriter doesn't seem to save spikes to file on anything other than channels 0 and 2. This is with the RHD2164 headstage left floating, with 64 single channel electrodes in the spike detector, and the spike viewer clearly showing events happening on all 64 chans. I don't actually use this version of BinaryWriter myself, but I wonder if this is expected...

(I use a greatly simplified BinaryWriter in my own branch, which detects spikes on all channels with the same floating headstage and the same signal chain)

I also tried reverting back to e3c12aa from May 16 before I started pushing BinaryWriter changes to the dev branch, and I get the same thing.

See the attached zip file containing the .xml settings file and the two short example `spike_electrode_indices.npy` files: [binary_spike_detect_chan_issue.zip](https://github.com/open-ephys/plugin-GUI/files/2146442/binary_spike_detect_chan_issue.zip)

aacuevas, 2018-06-29:
This is definitely a bug, and a serious one. For some reason, all channels except 0 are being registered as 2.
Seems to be some issue when generating the indices, I'll try to fix it as soon as possible.

mspacek, 2018-06-29:
The other thing to note is that the spike timestamps in `spike_times.npy` are very repetitive, i.e. lots of spike events have the exact same timestamp. Not sure if this is related to the channel issue, or if it just means that lots of channels tend to cross threshold simultaneously. Here's a full set of binary spike files: [spike_group_1.zip](https://github.com/open-ephys/plugin-GUI/files/2149656/spike_group_1.zip)

I also tried loading the waveforms in `spike_waveforms.npy` in python, and it looks like they're all unique, no repetitions. So that part seems to work. Here are 100 spikes. I think the threshold was -50 uV:
[image]

aacuevas, 2018-06-29:
If you are running it with a floating headstage that could be a reason, I've noticed that that kind of noise tends to sync quite well on all channels.

The cortex sample signal should give more realistic values.

aacuevas, 2018-06-30:
It should be fixed now. Spike channel numbers should report the appropriate values, with indexing starting from 1 as we do with other channel numbering in saved files.

I've also checked with the cortex sample file and timestamps do indeed vary.

beOn, 2018-07-04:
Quick favor to ask... would you mind referencing github issues in you commit comments when you know the commit addresses an issue? It makes it a lot easier to tell what branch users should pull to get the fixed code. I presume it's development, here, but will have to go check the commits to be sure. Thanks for all the hard work!

mspacek, 2018-07-04:
@beOn looks like it was fixed in 5785d0d. @aacuevas, I can confirm that the channel indices seem to be saved correctly now in the `spike_electrode_indices.npy` file.

closed 2018-07-04 by mspacek