open-ephys/plugin-GUI #243: Spikes lost between SpikeSorter and SpikeViewer

holmescharles, 2018-07-23:
This problem exists in the current master branch. 
I have set up a fake spiking neuron with a TTL line. If I fire off 50 spikes with 160 ms inter-spike interval, SpikeSorter will display all 50 while Spike Viewer will display (and write) fewer than 50 typically (usually on the order or 3-6 fewer).

**How to reproduce:**
1.  Add some diagnostic print statments to SpikeSorter.cpp and SpikeDisplayNode.cpp to help with the exact counting. I personally added a counter to each module that would update and print after each spike it sees. Examples forked from the master branch can be seen here: (https://github.com/holmescharles/plugin-GUI.git).
2. Set up some kind of fake spike source. I used a TTL line and a voltage splitter and fired spikes every 160 ms.
3. Set the threshold on SpikeSorter to only see your fake spikes. For me, they were large compared to the noise, so this was trivial. E.g.: 
[image]
4. Leave the SpikeViewer threshold at 0.
5. Run.

**Expectations:**
SpikeViewer should see each spike that SpikeSorter sees. So after a run of spikes, their counts should be the same. But instead, the SpikeViewer count will be less than the SpikeSorter count. E.g, with my diagnostics, I see:
[DIAG] Spikes (SpikeSorter) = 1
[DIAG] Spikes saved (SpikeViewer) = 1
[DIAG] Spikes (SpikeSorter) = 2
[DIAG] Spikes (SpikeSorter) = 3
[DIAG] Spikes saved (SpikeViewer) = 2
[DIAG] Spikes (SpikeSorter) = 4
[DIAG] Spikes saved (SpikeViewer) = 3
[DIAG] Spikes (SpikeSorter) = 5
[DIAG] Spikes saved (SpikeViewer) = 4
[DIAG] Spikes (SpikeSorter) = 6
[DIAG] Spikes (SpikeSorter) = 7
[DIAG] Spikes saved (SpikeViewer) = 5
[DIAG] Spikes (SpikeSorter) = 8
[DIAG] Spikes saved (SpikeViewer) = 6
[DIAG] Spikes (SpikeSorter) = 9
[DIAG] Spikes saved (SpikeViewer) = 7
[DIAG] Spikes (SpikeSorter) = 10
[DIAG] Spikes (SpikeSorter) = 11
[DIAG] Spikes saved (SpikeViewer) = 8
[DIAG] Spikes (SpikeSorter) = 12
[DIAG] Spikes saved (SpikeViewer) = 9
[DIAG] Spikes (SpikeSorter) = 13
[DIAG] Spikes saved (SpikeViewer) = 10
[DIAG] Spikes (SpikeSorter) = 14
[DIAG] Spikes saved (SpikeViewer) = 11
[DIAG] Spikes (SpikeSorter) = 15
[DIAG] Spikes saved (SpikeViewer) = 12
[DIAG] Spikes (SpikeSorter) = 16
[DIAG] Spikes saved (SpikeViewer) = 13
[DIAG] Spikes (SpikeSorter) = 17
[DIAG] Spikes saved (SpikeViewer) = 14
[DIAG] Spikes (SpikeSorter) = 18
[DIAG] Spikes saved (SpikeViewer) = 15
[DIAG] Spikes (SpikeSorter) = 19
[DIAG] Spikes saved (SpikeViewer) = 16
[DIAG] Spikes (SpikeSorter) = 20
[DIAG] Spikes saved (SpikeViewer) = 17
[DIAG] Spikes (SpikeSorter) = 21
[DIAG] Spikes saved (SpikeViewer) = 18
[DIAG] Spikes (SpikeSorter) = 22
[DIAG] Spikes saved (SpikeViewer) = 19
[DIAG] Spikes (SpikeSorter) = 23
[DIAG] Spikes saved (SpikeViewer) = 20
[DIAG] Spikes (SpikeSorter) = 24
[DIAG] Spikes saved (SpikeViewer) = 21
[DIAG] Spikes (SpikeSorter) = 25
[DIAG] Spikes saved (SpikeViewer) = 22
[DIAG] Spikes (SpikeSorter) = 26
[DIAG] Spikes saved (SpikeViewer) = 23
[DIAG] Spikes (SpikeSorter) = 27
[DIAG] Spikes saved (SpikeViewer) = 24
[DIAG] Spikes (SpikeSorter) = 28
[DIAG] Spikes saved (SpikeViewer) = 25
[DIAG] Spikes (SpikeSorter) = 29
[DIAG] Spikes saved (SpikeViewer) = 26
[DIAG] Spikes (SpikeSorter) = 30
[DIAG] Spikes saved (SpikeViewer) = 27
[DIAG] Spikes (SpikeSorter) = 31
[DIAG] Spikes saved (SpikeViewer) = 28
[DIAG] Spikes (SpikeSorter) = 32
[DIAG] Spikes saved (SpikeViewer) = 29
[DIAG] Spikes (SpikeSorter) = 33
[DIAG] Spikes saved (SpikeViewer) = 30
[DIAG] Spikes (SpikeSorter) = 34
[DIAG] Spikes saved (SpikeViewer) = 31
[DIAG] Spikes (SpikeSorter) = 35
[DIAG] Spikes saved (SpikeViewer) = 32
[DIAG] Spikes (SpikeSorter) = 36
[DIAG] Spikes saved (SpikeViewer) = 33
[DIAG] Spikes (SpikeSorter) = 37
[DIAG] Spikes saved (SpikeViewer) = 34
[DIAG] Spikes (SpikeSorter) = 38
[DIAG] Spikes saved (SpikeViewer) = 35
[DIAG] Spikes (SpikeSorter) = 39
[DIAG] Spikes saved (SpikeViewer) = 36
[DIAG] Spikes (SpikeSorter) = 40
[DIAG] Spikes saved (SpikeViewer) = 37
[DIAG] Spikes (SpikeSorter) = 41
[DIAG] Spikes saved (SpikeViewer) = 38
[DIAG] Spikes (SpikeSorter) = 42
[DIAG] Spikes saved (SpikeViewer) = 39
[DIAG] Spikes (SpikeSorter) = 43
[DIAG] Spikes saved (SpikeViewer) = 40
[DIAG] Spikes (SpikeSorter) = 44
[DIAG] Spikes saved (SpikeViewer) = 41
[DIAG] Spikes (SpikeSorter) = 45
[DIAG] Spikes saved (SpikeViewer) = 42
[DIAG] Spikes (SpikeSorter) = 46
[DIAG] Spikes saved (SpikeViewer) = 43
[DIAG] Spikes (SpikeSorter) = 47
[DIAG] Spikes saved (SpikeViewer) = 44
[DIAG] Spikes (SpikeSorter) = 48
[DIAG] Spikes saved (SpikeViewer) = 45
[DIAG] Spikes (SpikeSorter) = 49
[DIAG] Spikes saved (SpikeViewer) = 46
[DIAG] Spikes (SpikeSorter) = 50
[DIAG] Spikes saved (SpikeViewer) = 47

As you can see, in this particular run, SpikeViewer only saw 47/50 spikes!!

The TTL spikes may not be great for reproduction, but one should be able to reproduce this with any clear spikes in some signal. The key point is that EVERYTHING SpikeSorter sees should be passed along to SpikeViewer, but its not!

**POSSIBLY RELATED:** There are two thresholds displayed on SpikeViewer. One is the amplitude threshold that is unique to SpikeViewer. The other is a "re-displaying" of the SpikeSorter threshold. However, this seems to be mirrored about the zero line. E.g. see channel 17 (16 on SpikeViewer) in this plot: 
[image]

jsiegle, 2018-07-23:
Thanks a lot for doing this test—the Spike Viewer is indeed dropping spikes that are detected by either the Spike Sorter or Spike Detector. The cause is the buffering that's happening between callbacks: the Spike Sorter stops looking for spikes 50 samples before the end of one buffer, then adds those 50 samples to the start of the next buffer, so spikes that occur at the border won't have data points missing.

The problem arises when a spike is detected in the 50 samples that are carried over from the previous buffer. In this case, the spike has a `peakIndex` which is negative, because it peaks before the start of the current buffer. However, spikes with a negative `peakIndex` are ignored by the Spike Viewer. These are the spikes that are lost.

I was able to solve the problem by adding 50 to the `peakIndex` variable that gets passed to the `addSpike` function, so it won't be negative. Can you let me know if that fixes it for you?

holmescharles, 2018-07-23:
I made this change in my local copy and it seems to work! Is this something that can be added immediately to the development branch?

PS: What about my secondary issue regarding the mirrored thresholding? Any thoughts on that? Thank you!

jsiegle, 2018-07-23:
Let me run some tests to make sure there aren't any unintended consequences of shifting the peakIndex. If so, we should probably push a hotfix to the master branch, since this is a serious bug.

The mirrored thresholds are likely a consequence of having different conventions for the Spike Sorter and the Spike Display. This should be easy to fix.

beOn, 2018-07-24:
Thanks for the catch, Chuck.

aresjung, 2018-08-28:
What a great discovery!
I'd like to make a change to my local copy.
Could you please be more specific on where I should apply that change?

aacuevas, 2018-08-31:
The version on the testing branch should have this bug fixed

closed 2022-07-30 by jsiegle