open-ephys/plugin-GUI #245: [Feature request] Spike Sorter audio monitoring improvement

holmescharles, 2018-07-24:
**The current implementation** of audio monitoring in the Spike Sorter is cumbersome. Currently, to monitor a series of channels, one needs to advance the channel (either by using the button at the top of the window or using the drop down menu) and then click the audio button on the panel in the signal chain. This is a lot of mouse movement!

**A better implementation** would be to have a check box that causes the audio monitoring to follow you to what ever channel you switch to. The current functionality would be preserved, but if you wanted to observe many channel quickly, it would be easier.

I'd be willing to implement this if this idea is supported. Thanks!

jsiegle, 2018-07-28:
I agree that approach would be much more convenient! Let us know if you need any help implementing it.

jsiegle, 2022-07-30:
This hasn't been implemented exactly as described, but it's much easier to monitor different electrodes now. You can either:

- Select an electrode to listen to from the drop-down menu in the new [Audio Monitor](https://open-ephys.github.io/gui-docs/User-Manual/Plugins/Audio-Monitor.html) plugin
- Click the `MON` button in the upper-right corner of any electrode view in the [Spike Viewer](https://open-ephys.github.io/gui-docs/User-Manual/Plugins/Spike-Viewer.html) (assuming there's an Audio Monitor somewhere else in the signal chain)

closed 2022-07-30 by jsiegle