open-ephys/plugin-GUI #252: Group permissions don't match system default

holmescharles, 2018-08-17:
I've noticed that the default permissions of the binary files and such output by Open-Ephys do not match my user's default permissions.

My user's default permissions are "-rw-rw-r--", but the binary files output by Open-Ephys are "-rw-r--r--". The important part is that group permissions are read-only when they should be read/write. **Is there a way to set this to be read/write?**

For context, I am outputting raw binary files, not "open-ephys format" files.

aacuevas, 2018-08-17:
I've just checked and this stems from JUCE posix code which, for some reason, specifies a fixed 644 mode mask to open calls.

While possible to change, modifying JUCE code is always tricky for a number of maintenance reasons.
In the meanwhile, if you wish to try changing it yourself, the tricky part is in JuceLibraryCode/modules/juce_core/native/juce_posix_SharedCode.h line 513

holmescharles, 2018-08-17:
Thanks! I'm not sure I want to edit code to patch this issue locally.
I may just chmod the files immediately after open-ephys writes them.

closed 2022-07-30 by jsiegle