open-ephys/plugin-GUI #253: How to get first time stamp if timestamps.npy (flat binary) not written properly?

holmescharles, 2018-08-23:
**What I'm doing:** 
I am using a spike sorting program (_Spyking Circus_) to sort my spikes offline. After I get the spike times, I am using the sample numbers of the spikes (provided by _spyking circus_) and the time stamps numpy file (_timestamps.npy_, given with flat binary data) to get absolute spike times.

**Intermittent issue:** 
Normally this works fine, but I have a data session that has a truncated _timestamps.npy_ file. It is only 7337727 elements long while the raw data has 10142720 samples per channel!

**Insufficient fix:** 
As a work around, I thought I could use the sync time in _sync_messages.txt_, but occasionally this does not match the first element of _timestamps.npy_, so I am not sure its safe to use.

**_My question:_** 
**What is the best way to get the time stamp of the first sample of my ephys data if my timestamps.npy file is incomplete?** I _would_ just use the first value from _timestamps.npy), but I'm not sure if I can trust it if the file is incomplete.

jsiegle, 2018-08-23:
If the first or last timestamp is present, it's safe to interpolate the remaining samples. If you assume, for example, that the last 7337727 timestamps are all there, is there any way to check whether your alignment is correct (e.g., by looking at the response to some external stimulus)?

Also, was there anything different about this recording that could explain why the timestamps file was incomplete? It would obviously be good to figure out what happened, in case there's a bug that needs to be fixed.

holmescharles, 2018-08-23:
I should mention that there were two independent issues, occurring with different recordings.

1. In one case, the time stamps file had fewer data points than the raw data (accounting for number of channels, obviously). In this case, the starting timestamp did match the sync_messages.txt sync time.
2. In another case, the time stamps file had the same number of data points as the raw data, but the start time stamp was 1024 samples greater than the sync_messages.txt sync time.

In either case, I've decided to simply use the sync time from the sync_messages.txt, even if there is a discrepancy. At most it is 1024 samples off, which is ~1/30th of a ms at a 30kHz sampling rate, which I can live with.

Thanks!

aacuevas, 2018-08-23:
Which version of the GUI are you using?
There were a couple of issues related to the binary format and sync_messages that were fixed on the testing branch, so it would be interesting to know if these are the same ones or new ones.

In any case, as josh said, it would be good to know if the issue can be reproduced so we can fix it in case there is an ongoing bug.

holmescharles, 2018-08-23:
I'm on the development branch from July 5th.

jsiegle, 2022-07-30:
There were a number of changes in GUI version 0.6.0 intended to make writing timestamps and sync messages files more reliable. If you're still experiencing this problem with the latest version of the GUI, please re-open this issue.

closed 2022-07-30 by jsiegle