open-ephys/plugin-GUI #258: Change LFP Viewer settings defaults (for discussion)

ethanbb, 2018-09-19:
This is personal preference I suppose, but anecdotally at least one other person and I have found these default settings in the new LFP Viewer counterintuitive:

* Turning off all TTL events ("Why isn't the event showing up?")
* Turning on median offset correction ("Why is this test signal 0 instead of the DC value I was expecting?")

The median offset correction I can see being something you'd want to use most of the time with real signals, but until you realize it's there it can make for a pretty confusing debugging session. Intuitively I would expect the viewer to just show exactly the data that's being fed into it (without an offset) unless I told it otherwise. The TTL events being off doesn't make any sense to me - if I'm using a TTL event, I'd almost always want visual confirmation that it is being received or sent.

jsiegle, 2018-09-19:
I agree with this. These settings were changed for Neuropixels recordings, but they are definitely not ideal for the acquisition board. @aacuevas, can you make sure this gets updated before the next release?

ethanbb, 2018-09-19:
That would be great, thanks!

aacuevas, 2018-09-20:
It should be done. If you could, please check I didn't miss anything.
Thanks!

ethanbb, 2018-09-20:
Looks good, thanks again!

closed 2018-09-20 by ethanbb