open-ephys/plugin-GUI #261: Feature to prevent recording with low disk space

angelonc, 2018-10-01:
Thanks so much for providing such an awesome open source recording tool!

I quite like the live estimates of CPU and disk usage, and I was wondering if it would be feasible to add in a feature for the record button that would prevent recording if there is free disk space below some threshold value (which would be configurable). On a few occasions, I haven't noticed that the disk was full, and the gui allows recording with no warning, and I've lost those recordings.

vigji, 2022-09-30:
I think this would be a life saver for many! If I can, I'd upvote this feature request :)

Thank you guys for creating this amazing tool!!

jsiegle, 2022-09-30:
This feature is almost ready, we will add it in the next release!