open-ephys/plugin-GUI #268: Getting arbitrary metadata values is unnecessarily repetitive

ethanbb, 2018-11-15:
I'm working on a feature that involves getting each piece of metadata from an event (could be from anything, though) as the correct type and converting it to a string. I want to use a template to make a variable of the correct type to receive each metadata value, If the `MetaDataValue` or `MetaDataDescriptor` class had just one method that returned a value of template type "T", that would be easy, but since the `getValue` functions require you to pass in a variable of type "T", I have to do something like the below. Can this be considered in future work on these classes?

Excerpt:
```
        case MetaDataDescriptor::MetaDataTypes::INT8:
            success = sendMetaDataValue<int8>(valuePtr);
            break;

        case MetaDataDescriptor::MetaDataTypes::UINT8:
            success = sendMetaDataValue<uint8>(valuePtr);
            break;

        case MetaDataDescriptor::MetaDataTypes::INT16:
            success = sendMetaDataValue<int16>(valuePtr);
            break;

        case MetaDataDescriptor::MetaDataTypes::UINT16:
            success = sendMetaDataValue<uint16>(valuePtr);
            break;

        case MetaDataDescriptor::MetaDataTypes::INT32:
            success = sendMetaDataValue<int32>(valuePtr);
            break;

        case MetaDataDescriptor::MetaDataTypes::UINT32:
            success = sendMetaDataValue<uint32>(valuePtr);
            break;

        case MetaDataDescriptor::MetaDataTypes::INT64:
            success = sendMetaDataValue<int64>(valuePtr);
            break;

        case MetaDataDescriptor::MetaDataTypes::UINT64:
            success = sendMetaDataValue<uint64>(valuePtr);
            break;

        case MetaDataDescriptor::MetaDataTypes::FLOAT:
            success = sendMetaDataValue<float>(valuePtr);
            break;

        case MetaDataDescriptor::MetaDataTypes::DOUBLE:
            success = sendMetaDataValue<double>(valuePtr);
            break;

...

template <typename T>
bool EventBroadcaster::sendMetaDataValue(const MetaDataValue* valuePtr) const
{
    Array<T> data;
    valuePtr->getValue(data);
    ....
}
```

aacuevas, 2018-11-15:
Unfortunately, this behavior is intended, in part due to limitations of C++ templates. One of the design goals of metadata was to have very strong typing, so metadata of one type can only be assigned to a variable of the same.
This was specially designed for developers with little C++ experience, since from experience, not having strict type limitations forced in a language like C++ that is, by nature, strongly typed itself end being the source of many bugs and issues.
Of course, advanced developers can always work around the limitations, but we work under the assumption that such coders have a good understanding of the issues that can arise. Of course, this usually mean more code to work around our forced limitations.

Now, even with all this, ideally I would have preferred having something of the form
```
template<typename T>
T MetaDataValue::getValue() const
{ insert_type_checks_here }

int16 a = value1->getValue();
float b = value2->getValue();
```
Unfortunately, that would not compile. In C++ the return type is not part of a method signature, so the compiler does not know which method is being called. You would need to do something like
`int16 a = value1->getValue<int16>();`
Of course, having to manually insert the type defeats the purpose of having a template. It would also allow something like
`int16 b = value2->getValue<float>();`
Which is something we want to avoid.

We do offer a MetaDataValue::getRawValuePointer() for type-agnostic processors (like the binary writer does, which just takes the data and writes it to disk), but it generally not recommended.

Of course, any suggestion on how we could improve this interface, keeping in mind both the language limitations and the strong typing we want to achieve, would be appreciated.

ethanbb, 2018-11-15:
OK, thanks for the explanation and I appreciate the effort you put into reducing the likelihood of bugs for C++ newbies. I just tried coming up with an alternative solution, and I guess it's a bit more complicated than I initially thought to make a method with return type T that depends on the metadata type. I was thinking of Juce classes like `Array` that are themselves templates and can do this, but making a method of a non-template class like `MetaDataValue` behave similarly would be hard. So the way it is right now is probably best.

closed 2018-11-15 by ethanbb