open-ephys/plugin-GUI #274: Audio monitor merging channels

vncntprvst, 2018-12-12:
Hi,
I have a strange bug with the audio monitor (I know it's being revamped, so maybe there's already a fix to this). When monitoring a new channel, the audio monitor is doing what sounds like a mix of the former and the new signal. The only way to stop this is to stop the recording and click play again. Which of course is not an option when saving data. The audio monitor is already clunky (see e.g., https://github.com/open-ephys/plugin-GUI/issues/245), but this makes it really bad. 
I'm running a version of the plugin GUI compiled from source two months ago, on Windows 7 64 bit. 
Thanks

aacuevas, 2018-12-18:
I've fixed some issues with channel selection while acquisition is active, both for audio monitoring and recording. It should have fixed this, as well as other shenanigans, but please check with the latest version (any branch, in this case) if it fixes the issue for you.

vncntprvst, 2018-12-18:
Great. I'll check that asap.

vncntprvst, 2018-12-21:
I use the Monitor button in the Spike Sorter module to listen to channel activity. I use a configuration where I only record ADC signals (ie, not headstage), and with these recent updates (development branch), now there's no audio output at all! I should test with a headstage to compare. But if this could be fixed, I would really appreciate. Thanks!

vncntprvst, 2018-12-21:
I check the audio for the headstage signals: it works. Audio for the ADC signal works in the Audio tab selector up to the Splitter, but not after, even though the signal is passed along. That's a bit weird, but I can live with that for now.

closed 2020-12-15 by medengineer