open-ephys/plugin-GUI #287: Online Tuning Curves

angelonc, 2019-01-26:
I'm interested in setting up online tuning curve / laser power response plots in the openEphys GUI, but I'm a bit at a loss of where to start.

I know I should send trial information from matlab via network events, but the documentation seems somewhat sparse as to how to set up, say the average spike count for each condition triggered by a ttl pulse for that trial. On the [PSTH page](https://open-ephys.atlassian.net/wiki/spaces/OEW/pages/23265293/PSTH), I think this section describes what I want to do, but are there any more specific examples of how to do this:

```
** High precision real time trial alignment

Experiments requiring high temporal precision in trial alignment can be accomplished by sending trial start & trial type information over TCP/IP, but trial align via a TTL pulse on one of the digital input lines.

This needs to be setup in the PSTH GUI accordingly. 

An example how to send string messages to GUI from a c++ / matlab session will be available on github.
```

Any pointers would be much appreciated!

jsiegle, 2019-01-28:
We haven't been maintaining the PSTH module, but the [Event Triggered Average](https://open-ephys.atlassian.net/wiki/spaces/OEW/pages/129073153/Event+Triggered+Averaging+EvntTrigAvg) module has similar functionality.

Each Event Triggered Average module is triggered by a single TTL input line, but you can place several modules together in a signal chain to monitor the responses to a range of conditions. Does that seem like it would do what you need it to?

jsiegle, 2019-01-28:
Yeah, that's the basic idea. So right now you'd be limited to 8 different conditions. We could bump it up to 256 by using the full TTL "word" to define each condition, but that's a lot of separate windows to have open at once.

I think a better option would be to modify the Event Triggered Average source code so each channel holds the response of one electrode to a different TTL trigger, rather than the response of a different electrode to the same TTL trigger. @Claybarn, does that sound doable?

Claybarn, 2019-01-28:
Don't think this change would be too hard to implement. Would you then need an Event Triggered Averaging module for each electrode? That could become cumbersome depending on the number of electrodes. A general solution would be to collect spikes in reference to every TTL and have a drop down menu by each graph to select what alignment to view, but that would be a lengthier project. 

@angelonc do you want more than 8 tuning curves? If so you could check out the python plugin. It is easy to communicate over ports and could easily average spikes within a given time window after receiving an event. Wouldn't be limited to the number of tuning curves there. 

Happy to assist with either project!

angelonc, 2019-01-29:
@jsiegle and @Claybarn, thanks for your quick replies!

We do have some pretty high channel counts, up to 256, so it sounds like hacking the Event Triggered Average may be too cumbersome.

At the moment, we really only want 1 tuning curve, but it would have 20-50 conditions (in our case, auditory tone frequencies) plotted within it. The Python plugin looks promising, but seems to currently only work on Linux and MacOS (unfortunately our acquisition system is on a Windows PC). I'll have to think about whether it's worth switching at this stage.

Is there any possibility of streaming spike times and event data to another PC, and then use a programming environment there to do the online analysis? Maybe via the Event Broadcaster?

Claybarn, 2019-01-29:
Unfortunately the python plugin is not yet supported for windows, but that is under active development. 

In the meantime I think you could stream data back and forth from another PC by using the event broadcaster and network events modules. I have less experience with these modules, so @jsiegle might know if this is feasible better than I would.

jsiegle, 2019-02-03:
That's right, you can use the EventBroadcaster to send spikes + events to another computer. There's no documentation page for it currently, so you'll have to ask the forum to find out if someone already has code for interfacing with it. Getting some documentation / tutorials for this module is on our to-do list.

Claybarn, 2019-02-03:
If you can’t find any information on the forum I can help you out.

angelonc, 2019-02-12:
Cool, thanks! I'll DM you

closed 2020-12-15 by medengineer