open-ephys/plugin-GUI #288: Failed to load winusb.dll, okFrontLib.dll, and RhythmNode.dll

alejoe91, 2019-01-29:
Hi guys,

I'm compiling the master branch of the GUI using Windows 10 and VS 2013.
The building process goes smooth, but when I start the open-ephys.exe it fails to load the RhythmNode and IntanRecordingController.

I have installed the OpalKelly FrontPanel USB Driver 4.4.0, but this happens even when the acquisition bord is not plugged in.

The output form VS is:

```
'open-ephys.exe' (Win32): Loaded 'C:\apps\open-ephys-plugin-GUI\Builds\VisualStudio2013\x64\Debug64\bin\plugins\IntanRecordingController.dll'. Symbols loaded.
'open-ephys.exe' (Win32): Loaded 'C:\apps\open-ephys-plugin-GUI\Builds\VisualStudio2013\x64\Debug64\bin\okFrontPanel.dll'. Module was built without symbols.
'open-ephys.exe' (Win32): Loaded 'C:\Windows\System32\winusb.dll'. Symbols loaded.
'open-ephys.exe' (Win32): Unloaded 'C:\Windows\System32\winusb.dll'
'open-ephys.exe' (Win32): Unloaded 'C:\apps\open-ephys-plugin-GUI\Builds\VisualStudio2013\x64\Debug64\bin\okFrontPanel.dll'
'open-ephys.exe' (Win32): Unloaded 'C:\apps\open-ephys-plugin-GUI\Builds\VisualStudio2013\x64\Debug64\bin\plugins\IntanRecordingController.dll'
```

and 

```
'open-ephys.exe' (Win32): Loaded 'C:\apps\open-ephys-plugin-GUI\Builds\VisualStudio2013\x64\Debug64\bin\plugins\RhythmNode.dll'. Symbols loaded.
'open-ephys.exe' (Win32): Loaded 'C:\apps\open-ephys-plugin-GUI\Builds\VisualStudio2013\x64\Debug64\bin\okFrontPanel.dll'. Module was built without symbols.
'open-ephys.exe' (Win32): Loaded 'C:\Windows\System32\winusb.dll'. Symbols loaded.
'open-ephys.exe' (Win32): Unloaded 'C:\Windows\System32\winusb.dll'
'open-ephys.exe' (Win32): Unloaded 'C:\apps\open-ephys-plugin-GUI\Builds\VisualStudio2013\x64\Debug64\bin\okFrontPanel.dll'
'open-ephys.exe' (Win32): Unloaded 'C:\apps\open-ephys-plugin-GUI\Builds\VisualStudio2013\x64\Debug64\bin\plugins\RhythmNode.dll'
```

I also tried to move the `winusb.dll` in the bin folder, with no luck.

Has anyone experienced this problem before? Whn I run the binaries from the `windows-64` repo the Rhythm FPGA plugin is loaded correctly and the acquisition board works fine.

Thanks!

Alessio

aacuevas, 2019-01-30:
Hi Alessio,

First, the repo you said you are getting the binaries from is outdated. The official repo, which is linked in both the webpage and the wiki, is this one: https://github.com/open-ephys-GUI-binaries/open-ephys/tree/windows

Regarding the issue, those are normal messages from visual studio and do not indicate any particular problem, neither with the plugins or the winusb.dll file, that just happens to be a dll the front panel library uses.

The FrontPanel dll does need the Visual C++ 2015 redistributable package, which you can download from here https://www.microsoft.com/en-US/download/details.aspx?id=52685

If that did not solve your issue, please run the GUI with the --console command line argument and send me the contents of it, in case it had more clues.

Aarón

alejoe91, 2019-02-07:
Hi @aacuevas!

Installing the Visual C++ 2015 inded solves the issue :)

Thanks
Alessio

closed 2019-02-07 by alejoe91