open-ephys/plugin-GUI #292: Documentation errors and failure in building from the source

kouichi-c-nakamura, 2019-02-26:
I've tried to build Open Ephys Plugin-GUI and failed. Also I found lots of small errors in the documentation.

## The "our repository" link

[This page](http://www.open-ephys.org/gui) says that it's better to use GitHub and build the GUI in order to keep updated. The "our repository" link on the page brought me to https://github.com/open-ephys/GUI/, which in turn says that this is not active any more and use 

https://github.com/open-ephys/plugin-GUI instead. With that, I think that the "our repository" link should point to https://github.com/open-ephys/plugin-GUI from the beginning to avoid confusion.

## File/folder paths for plugin-GUI

The readme file of https://github.com/open-ephys/plugin-GUI then brings you to [this page](https://open-ephys.atlassian.net/wiki/spaces/OEW/pages/491621/Windows) if you are to opt for building from the source.

[The "Building from source" documentation](https://open-ephys.atlassian.net/wiki/spaces/OEW/pages/491621/Windows#Windows-Buildingfromsource) also contains a lot of confusions.

In list item 2, it refers to the repo [https://github.com/yourusername/plugin-GUI](https://github.com/yourusername/plugin-GUI.git), but later when it mentions file/folder paths, all of them are wrong in some way, though it is not difficult to guess what it meant.

1. In list item 7, it refers to `GUI/Resources/DLLs/FrontPanelUSB-DriverOnly-4.4.0.exe`.  But I think it should be`plugin-GUI\Resources\FrontPanelUSB-DriverOnly-4.5.5.exe`
2. In list item 8, it refers to `GUI/Builds/VisualStudio2013`, where it should be `plugin-GUI/Builds/VisualStudio2013`
3. In list item 10, it refers to `GUI/Resources/Bitfiles/rhd2000.bit` , where it should be `plugin-GUI/Resources/Bitfiles/rhd2000.bit`
4. In list item 10, it refers to `Debug/bin folder`, where it should be `plugin-GUI\Builds\VisualStudio2013\Debug\bin`

Moreover, in the [Building the plugins](https://open-ephys.atlassian.net/wiki/spaces/OEW/pages/491621/Windows#Windows-Buildingtheplugins) section, `Builds\Visualstudio2013\Plugins\Plugins.sln` should be `plugin-GUI\Builds\Visualstudio2013\Plugins\Plugins.sln` for clarity.

## Errors in building plugins

So I managed to build the plugin GUI, but I encountered 15 errors and 48265 warnings while building the plugins subsequently. I'm using **Windows 10** and **Visual Studio Community 2013** (12.0.31101.00 update 4). As shown below, 13 out of 15 errors were about "cannot open input file 'open-ephys.lib'", which I can find at `plugin-GUI\Builds\VisualStudio2013\Debug\bin`

I've also tried the following in the Command Prompt, only to get an error:

> The module C:\Program Files (x86)\EcubeAPI\ecubeapi32.dll " failed to load. Make sure the binary is stored at the specified path or debug it to check for problems with the binary or dependent .DLL files. The specified module could not be found..

```
cd \Windows\SysWOW64
regSvr32 "C:\Program Files (x86)\EcubeAPI\ecubeapi32.dll"
regSvr32 "C:\Program Files (x86)\EcubeAPI\ecubeapi64.dll"
```

I can confirm that there is no `C:\Program Files (x86)\EcubeAPI` folder.

#134 might be related, but not sure.

| Error | id    | Description                                                  | File                                                         | Project           |
| ----- | ----- | ------------------------------------------------------------ | ------------------------------------------------------------ | ----------------- |
| Error | 48280 | error LNK1181: cannot open input file   'DspLib.lib'         | C:\Users\xxxxxx\Documents\plugin-GUI\Builds\VisualStudio2013\Plugins\FilterNode\LINK | FilterNode        |
| Error | 6073  | error LNK1181: cannot open input file 'open-ephys.lib'       | C:\Users\xxxxxx\Documents\plugin-GUI\Builds\VisualStudio2013\Plugins\CommonAvgRef\LINK | CommonAvgRef      |
| Error | 7203  | error LNK1181: cannot open input file 'open-ephys.lib'       | C:\Users\xxxxxx\Documents\plugin-GUI\Builds\VisualStudio2013\Plugins\ChannelMapper\LINK | ChannelMapper     |
| Error | 12750 | error LNK1181: cannot open input file 'open-ephys.lib'       | C:\Users\xxxxxx\Documents\plugin-GUI\Builds\VisualStudio2013\Plugins\NetworkEvents\LINK | NetworkEvents     |
| Error | 13164 | error LNK1181: cannot open input file 'open-ephys.lib'       | C:\Users\xxxxxx\Documents\plugin-GUI\Builds\VisualStudio2013\Plugins\Phase   Detector\LINK | Phase Detector    |
| Error | 13460 | error LNK1181: cannot open input file 'open-ephys.lib'       | C:\Users\xxxxxx\Documents\plugin-GUI\Builds\VisualStudio2013\Plugins\SerialInput\LINK | SerialInput       |
| Error | 14782 | error LNK1181: cannot open input file 'open-ephys.lib'       | C:\Users\xxxxxx\Documents\plugin-GUI\Builds\VisualStudio2013\Plugins\BasicSpikeDisplay\LINK | BasicSpikeDisplay |
| Error | 20168 | error LNK1181: cannot open input file 'open-ephys.lib'       | C:\Users\xxxxxx\Documents\plugin-GUI\Builds\VisualStudio2013\Plugins\PulsePalOutput\LINK | PulsePalOutput    |
| Error | 22363 | error LNK1181: cannot open input file 'open-ephys.lib'       | C:\Users\xxxxxx\Documents\plugin-GUI\Builds\VisualStudio2013\Plugins\SpikeSorter\LINK | SpikeSorter       |
| Error | 23601 | error LNK1181: cannot open input file 'open-ephys.lib'       | C:\Users\xxxxxx\Documents\plugin-GUI\Builds\VisualStudio2013\Plugins\RecordControl\LINK | RecordControl     |
| Error | 25986 | error LNK1181: cannot open input file 'open-ephys.lib'       | C:\Users\xxxxxx\Documents\plugin-GUI\Builds\VisualStudio2013\Plugins\Rectifier\LINK | Rectifier         |
| Error | 27601 | error LNK1181: cannot open input file 'open-ephys.lib'       | C:\Users\xxxxxx\Documents\plugin-GUI\Builds\VisualStudio2013\Plugins\ArduinoOutput\LINK | ArduinoOutput     |
| Error | 30429 | error LNK1181: cannot open input file 'open-ephys.lib'       | C:\Users\xxxxxx\Documents\plugin-GUI\Builds\VisualStudio2013\Plugins\EventBroadcaster\LINK | EventBroadcaster  |
| Error | 32039 | error LNK1181: cannot open input file 'open-ephys.lib'       | C:\Users\xxxxxx\Documents\plugin-GUI\Builds\VisualStudio2013\Plugins\BinaryWriter\LINK | BinaryWriter      |
| Error | 25177 | error LNK1181: cannot open input file 'OpenEphysHDF5Lib.lib' | C:\Users\xxxxxx\Documents\plugin-GUI\Builds\VisualStudio2013\Plugins\KWIKFormat\LINK | KWIKFormat        |

Can somebody fix those errors in documentation? And also can you give me an advice on how to build it from the source? Have I missed something?

MMathisLab, 2019-02-26:
Possibly related, possibly not, but are you using Anaconda to install? Evidently there was a pathe error recently so lots of users had been experiencing the “DLL ... module not found error” seems installing “conda install msvc_runtime” is fixing  it for most. 

https://forum.image.sc/t/tensorflow-dlc-2-0-importerror-dll-load-failed-the-specified-module-could-not-be-found/22991

kouichi-c-nakamura, 2019-02-26:
Thanks. I've tried this on two PCs; one certainly has Anaconda installed, while I'm not sure about the other one. I'll check that soon.

aacuevas, 2019-02-26:
Hi,
I'll fix the contradictions in the wiki. I'm probably going to remove the GUI/ parts of the paths, since almost everywhere else we just use folder paths relative to the root (be it GUI. plugin-GUI  or however the user wants it renamed)

Regarding your building errors. First, forget about the ecube stuff. That is a very specific plugin that is not even being built by default. I'll probably delete the references to it in the wiki too.

Have you checked that the build options for the GUI and the plugins match? You seem to be building the GUI with Debug|x86 so be sure that the plugins are set to the same (that said, we recommend building both the GUI and plugins for x64)

jsiegle, 2022-07-30:
All developer documentation now lives here: https://open-ephys.github.io/gui-docs/Developer-Guide/index.html

Please open an issue in the [`gui-docs`](https://github.com/open-ephys/gui-docs) repository if you find any mistakes.

closed 2022-07-30 by jsiegle