open-ephys/plugin-GUI #293: Spikes are dropped when writing to binary format's numpy files.

holmescharles, 2019-02-27:
My experimental setup uses in house software to run behavior with my subjects. In the past, we have directly sent spikes from other data acquisition software, e.g., Plexon, directly to this software so we and utilized it there (e.g., making PSTHs, etc). When using Open-Ephys (OE), the best solution we have is to read from the spike_times.npy and spike_electrode_indices.npy. However, not all the spikes seem to be written to this file!

**How I verified this**
1. I set up a fake spike generator. The actual implementation is not important, but the end result is I send impulses into OE. The signal-to-noise ratio is essentially infinite.
2. I set up my threshold on an arbitrary channel (lets say channel 7) on the SpikeSorter so that I will detect these spikes. I also raise the thresholds on all other channels so that they do not detect anything! (This is important because with my setup, spikes "bleed" into every channel).
3. I send 40 spikes, spaces by ~400 ms.
4. I run a simple python script  to check spike_electrode_indices.npy for all cases of that channel index (in our example, channel 7) and print the total.

The script prints 40.

But then, if I lower the threshold on another channel (lets say, channel 6) so that noise gets picked up, i.e., that channel gets >10,000 spikes, and I rerun by script, I only see ~29 spikes. The npy file does contain ~10,000 spikes in it, but I cannot be sure what percentage of the new channel's spikes are getting through.
Then, if I lower the threshold on ALL other channels (in our example, channels 1-6 and 8-32), and re-run my script, I only see ~6 spikes!

**So what's going on?**
1. Why are any spikes being dropped?
2. Why do more spikes get dropped as more and more spikes are written?

aacuevas, 2019-02-28:
This seems like a buffer issue. We've known of case where too many events in a short time could cause issues in some situations. We fixed some but this might be related,

Does this issue happen the same with both the spike sorter and the spike detector?
It is particular to the binary format or does it happen with other file formats as well?

holmescharles, 2019-03-02:
I have only noticed the spikes missing from the numpy file.
I have not noticed spikes missing from the Spike Sorter nor the Spike Display. If I generate 40 spikes, I see 40 spikes in the Spike Display. The Spike Display is downstream of the Spike Detector, therefore the Spike Detector must see the same 40 spikes.

I am building from source. I forked from the most recent version of the master branch. The last commit was d227ee12c36b990bc121fe244f9d123338c5b23d. Hope that helps!

aacuevas, 2019-03-05:
The detector or the sorter are the ones that "detect" the spikes and send them down the stream, while the display is the one that displays them as well as sends the spikes for recording, but only those that are visualized, i.e., those who pass the second threshold set at the display.

In your case which you have ~1000 spikes in the npy files, how are you verifying that they are shown?

I was asking if spikes are missing from the npy files whether you use the detector or the sorter. Also, if you could try with other file formats (open ephys, nwb, etc...) and check if spikes are also missing on those formats or just the binary, so we could pinpoint the issue.

Best,
Aarón

holmescharles, 2019-03-08:
> In your case which you have ~1000 spikes in the npy files, how are you verifying that they are shown?

I have a couple lines in the SpikeDisplayNode.cpp. I think its easiest to just show the edits here (starting with line 241), since there are so few (edits are indicated by DIAG):

```
if (aboveThreshold)
    {
        static int spike_count_DIAG = 0; // DIAG CDH
        // save spike
        if (isRecording)
        {
            // DIAG CDH
            if (electrodeNum == 6) // 0-index
                std::cout << "[DIAG] Spikes saved count = " << ++spike_count_DIAG << std::endl;
            // end DIAG
            CoreServices::RecordNode::writeSpike(newSpike, spikeInfo);
        }
        // add to buffer
        if (e->currentSpikeIndex < displayBufferSize)
        {
            //This releases the spike from the smart pointer to avoid copies, so it's done latest.
            e->mostRecentSpikes.set(e->currentSpikeIndex, newSpike.release());
            e->currentSpikeIndex++;
        }

    }
```

So basically, I generate X spikes artificially, and verify that my printed count increments by X.

> I was asking if spikes are missing from the npy files whether you use the detector or the sorter. Also, if you could try with other file formats (open ephys, nwb, etc...) and check if spikes are also missing on those formats or just the binary, so we could pinpoint the issue.

Ah. Apologies! I was using the Sorter, not the Detector. I have not tried that, but I can. I can also try the other file formats, but it might take a little bit of time to get back to you. I'll do my best!

holmescharles, 2019-03-11:
I just checked a few things.

1. When I substitute the Spike Detector for the spike viewer, the problem persists. The output numpy file has <25% of the expected number of spikes.
2. When I switch from the binary format to OpenEphys format, all the spikes appear to be in the output ".spikes" file.

aacuevas, 2019-03-12:
Thank you. Knowing that in open ephys format there is no spike drop is really useful, as it isolates the issue to the binary format itself (in fact, my first guess was elsewhere).
Another question, in a case where spikes have dropped, do all the .npy files have the same entry counts? Or is there a mismatch?
There should be the same number of entries on the channel_indexes, timestamps and waveforms npy files, but I want to be sure if that is the case in the case you describe.

Thanks,
Aarón

holmescharles, 2019-03-12:
They all have the same number of spikes.

mspacek, 2019-03-27:
Not sure if this is relevant, but I remember adding some code a while ago that would only write to the .npy files when the number of events in buffer exceeded a threshold. I think this was to prevent high CPU usage. So the number of spikes or perhaps digital input events at any point in time actually flushed to a .npy file would always be less than the total number. However, at the end of the recording, or the next time the buffer hit its threshold, all remaining events are written to the .npy file.

Here's the buffer size definition:
```cpp
        // flush file buffer to disk and update the .npy header every this many records:
        const int recordBufferSize{ 1024 };
```
https://github.com/open-ephys/plugin-GUI/blob/master/Source/Plugins/BinaryWriter/NpyFile.h#L71

And here's where it's used:
https://github.com/open-ephys/plugin-GUI/blob/master/Source/Plugins/BinaryWriter/NpyFile.cpp#L195

aacuevas, 2019-03-27:
That brings an important point... Could you send (to aaron@open-ephys.org) a npy file which has skipped spikes, preferably a indices or times one (for size)?
It might be that the data is there but the npy header has not been properly updated. I'd like to check that.

holmescharles, 2019-04-05:
Apologies for the delay. Sending now.

aacuevas, 2019-05-21:
Hi,

An update about this. I have been testing with my own setup and I am unable to reproduce this on my machine.

When you set the thresholds, so one channel gets the spikes and the others get noise or not, do you adjust them in the sorter/detector or in the viewer?

holmescharles, 2019-05-30:
I adjust them on the detector, though I can adjust them on the viewer and get the same result.
To be clear, both modules must have (close to) zero thresholds for the spikes to get through, so in a way, the answer is "both".

If you are having trouble fixing it, I wouldn't worry too much. We actually have gotten the EventBroadcaster to work well with our in-house C code, so this issue is no longer critical. That being said, I am happy to help resolve it to improve the OE project!

jsiegle, 2022-07-30:
We've tested this extensively with the latest version of the GUI and have not seen any evidence for dropped spikes. If you notice this in any of your recordings, please open a new issue.

closed 2022-07-30 by jsiegle