open-ephys/plugin-GUI #294: How can you send spikes to another piece of software?

holmescharles, 2019-02-27:
(I just posted another issue with the same introduction/background)

My experimental setup uses in house software to run behavior with my subjects. In the past, we have directly sent spikes from other data acquisition software, e.g., Plexon, directly to this software so we and utilized it there (e.g., making PSTHs, etc). When using Open-Ephys (OE), the best solution we have is to read from the spike_times.npy and spike_electrode_indices.npy. However, not all the spikes seem to be written to this file, so this does not seem to be a satisfactory solution. Additonally, I believe that the spikes are written in chunks. We read the spikes in at the end of every trial, but spikes will be missing/truncated if the last set of spikes has not been written yet.

Is there a better way to send spikes to our in-house software? It seems that ZeroMQ might be able to handle this, but I cannot find a plugin that does what I need. Any pointers?

aacuevas, 2019-02-28:
Trying to read the files while recording is active will indeed not get immediate results. All files are updated in blocks for performance reasons. Additionally, even if the blocks have been written, the npy headers are updated even more sparsely, as it needs to be updated to reflect the number of entries currently written but is a relatively performance-expensive process when writing.

You might want to try the Event Broadcaster. This plugin send events and spikes via ZeroMQ in their raw form. You'd need to decode the messages, but you can find the format described in Sources/Events/Events.h

Claybarn, 2019-02-28:
If your in-house software is in python, you should check out this project: https://github.com/MemDynLab/PythonPlugin/tree/event_reciever

holmescharles, 2019-03-02:
@aacuevas, to be clear, are you referring to this plugin when you mention the Event Broadcaster?
https://github.com/MemDynLab/ZMQInterface

aacuevas, 2019-03-05:
No, I am referring to the EventBroadcaster plugin present in the main GUI repository.
The plugin you link seems to be coded for an older version of the GUI and won't compile with the current one.
Of course, you can also try clayton's plugin for direct python integration.

holmescharles, 2019-03-09:
Got it. Found the plugin. I was thrown because the plugin does not have documentation on the wiki and is not listed as a "Sink" under "Individual Modules". Do you or does anyone have a link to some documentation for this module? I plan to google around myself, but if anyone knows a good resource off the top of their head that would be very much appreciated. Thanks!

aacuevas, 2019-03-11:
Unfortunately that module is not well documented tight now.
What is does is create a listening socket over the network using the ZeroMQ messaging library, accept any connection from a client using said library and then send every event and spike received, under the format:
uint16 event type, double timestamps_in_seconds and then the raw message
You can find information about the message format in the GUI source code, specifically the file Processors/Events/Events.h

ZeroMQ has bindings for C, we are using the C++bindings ourselves.

jsiegle, 2022-07-30:
The easiest way to do this is with the **Event Broadcaster** plugin, which is now more thoroughly documented [here](https://open-ephys.github.io/gui-docs/User-Manual/Plugins/Event-Broadcaster.html). Please open a new issue in the [Event Broadcaster](https://github.com/open-ephys-plugins/event-broadcaster) repository if you have any questions about how to use it.

closed 2022-07-30 by jsiegle