open-ephys/plugin-GUI #295: Can we make the Dsp library in the main GUI PLUGIN_API?

ethanbb, 2019-03-26:
Vinnie Falco's [DSP filter library](https://github.com/vinniefalco/DSPFilters) is currently both a "common lib" for plugins (see #146) and included in the main GUI code under "Processors" for use in the `AudioNode`. This is working but the duplication seems unnecessary. Can we just add `PLUGIN_API` declarations to the version in the main GUI so that it is exported in the `open-ephys.lib` file (on Windows)? The places we would need to do that should be the same as where `COMMON_LIB` is currently declared in the common lib version. Then that common lib could be removed and we would add a new header file in `Source/Plugins/Headers` to access the DSP library.

aacuevas, 2019-03-27:
I don't see any reason why not. Not completely sure why we didn't do that in the first place.

ethanbb, 2019-03-28:
OK, great. I didn't realize it was also part of the main GUI way back when we were talking about how to use the DSP library (or libraries in general) in multiple different plugins.

PR #296 (2019-05-05): Use DSP library from main GUI instead of a common library plugin
Closes #295. Just deletes the `DspLib` plugin, adds a header to import the DSP library from the main GUI and `PLUGIN_API` declarations for Windows, and updates build files.

I don't have a Mac, so someone should verify that the modified OSX build files are correct. I just deleted files specific to `DspLib`, and then went through general and `FilterNode` files and deleted any references to it (comparing with the `Rectifier` project files for reference).
closed 2019-05-06 by ethanbb