open-ephys/plugin-GUI #298: second channel mapper does not load configuration properly

yagui, 2019-05-08:
I'm currently using development branch.

When a use two different channel mappers to record channels in different processors, there is always one channel mapper that does not load the recording channels, so I have to manually set them every time I load the configuration.

I created a small example to be load (using file reader as source)
[test.txt](https://github.com/open-ephys/plugin-GUI/files/3157778/test.txt)

In the file, both channel map in lines 73 and 157 have their channels set to record, but when this configuration is loaded, only one of them is actually setting them to record.

aacuevas, 2019-05-21:
Should be fixed now. It wasn't actually the second mapper, but any mapper that had the first channels disabled.

yagui, 2019-05-21:
Yes, it's working great now. Thanks!

closed 2019-05-21 by yagui