open-ephys/plugin-GUI #301: Timeout after playing (Linux Ubuntu 18.04)

Niboshi1, 2019-05-28:
Installed GUI from precompiled binary for Linux, and followed instructions [here](https://open-ephys.atlassian.net/wiki/spaces/OEW/pages/491546/Linux). I managed to successfully run the GUI, but when I start recording, it stops receiving LFP signals and freezes. 

 In terminal I get this error
```Enabling processors...
Source node received enable signal
Expecting 35 channels.
Number of 16-bit words in FIFO: 0
Is eval board running: 0
Starting acquisition.
Flushing FIFO.
Expecting blocksize of 13312 for 1 streams
Resizing buffer. Samples: 20000, Inputs: 35, event channels: 1
   Enabling VisualizerEditor
Beginning animation.
Filter Viewport disabled.

Adding audio callback.
DAC
CRITICAL: Timeout on pipe read. Check block and buffer sizes.
CRITICAL: Timeout on pipe read. Check block and buffer sizes.
CRITICAL: Timeout on pipe read. Check block and buffer sizes.
CRITICAL: Timeout on pipe read. Check block and buffer sizes.
CRITICAL: Timeout on pipe read. Check block and buffer sizes.
```
[image]

I've checked issue [#159](https://github.com/open-ephys/plugin-GUI/issues/159), and made sure that acquisition board is plugged into USB3 port, but keep getting this error. Is there something I can do to fix this problem? Thanks.

aacuevas, 2019-05-28:
Hi,

It seems indeed transmission related, but just 32channels should not cause an issue like that. Let me ask some questions to understand the symptoms:

Does it happen when recording or just by acquiring?
Is it instantaneous or takes a while?
By the looks of your screenshot you could acquire for a couple of minutes before triggering the error, is that right?

Does it happen with just the Rythm node and a LFP viewer, instead of the full chain?
I see you are acquiring at 2KS/s, does the error occur when acquiring at different rates? (both slower and faster, I'd say try at 1, 10, 20 and 30 KS/s)

Finally, what are your computer specs?

Best,
Aarón

Niboshi1, 2019-05-28:
Hi Aarón
Thank you for the quick reply.

>Does it happen when recording or just by acquiring?

It happens when I start acquiring. That is, when I click the start button (▷). However, I've seen this happen after clicking the record button.
>Is it instantaneous or takes a while?

It depends. Sometimes, it's usually after a few minutes of recording as you pointed out, sometimes it's longer and freezes after 20 minutes of acquiring. There was one time without any error, but nothing was different (hardware or software-wise) from other times.
>Does it happen with just the Rythm node and a LFP viewer, instead of the full chain?

Yes, it happens with simpler chain. The time it takes to trigger the error does not seem to change.
>I see you are acquiring at 2KS/s, does the error occur when acquiring at different rates?

In my experience, acquisition with higher sampling rate, such as 30KS/s causes the error to be triggered quicker than with say 1KS/s. Last 3 times I sampled at 30KS/s, I got the error after 2-3 seconds. Maybe it's a trafficking problem?
>Finally, what are your computer specs?

CPU I'm using is AMD Ryzen Threadripper 2990WX 32-Core Processor, with 128GB memory.
I don't think it matters much but I also have two GPUs, Geforce 2080 ti

I hope these helps.

Niboshi1, 2019-05-28:
Forgot to add information on SSD. The one I'm using is Crucial 2.5' SSD 480GB

jsiegle, 2022-07-30:
The plugin for interfacing with the acquisition board now lives here: https://github.com/open-ephys-plugins/rhythm-plugins

If you're still experiencing this problem with the latest version of the GUI, please open an issue in that repository.

closed 2022-07-30 by jsiegle