open-ephys/plugin-GUI #311: Adding a new source disable channel mapper recordings

yagui, 2019-07-28:
I have two channel mappers which I use as recording nodes.
One channel mapper has the neural channels and the other has the accelerometer channels.
In the issue #298 there was a problem that every time I opened the configuration file the channel mapper that had the first channels disabled (the acceleromenter one), didn't load the configuration to record those channels.
That issue was fixed in #298 but now I find that after loading the configuration file, if I add a new source, the recording option in that channel mapper gets disabled. 
If I reenabled them, save the file with the new source (I'm testing it with the frame grabber plugin), and then open again the configuration file, it stays enabled. So the issue is only when I add a new source.

aacuevas, 2019-08-19:
Does that happen only when loading a chain, or if you have a mapper and add a new source, do the recorded channels also get disabled?

How are you adding that new source, with a merger?

yagui, 2020-01-25:
It also happens when I add a new filter, like a Bandpass filter or the CAR filter.
When I load the chain, previously configured with two channel mappers set for recording, everything is ok.
But if later I add a new filter or a new source, there is one channel mapper that disables recording. Enabling recording in the second channel mapper and saving the configuration is enough.

jsiegle, 2022-07-30:
This issue is no longer relevant as of GUI version 0.5.0 – now recording takes place exclusively inside Record Nodes.

closed 2022-07-30 by jsiegle