open-ephys/plugin-GUI #312: Randomly X_ConfigureWindow Error

reinderdorman, 2019-07-30:
I have noticed a few times the GUI has problems with being a specific size (Literally the size it is on the monitor). its provided the following error:

ERROR: X returned BadValue (integer parameter out of range for operation) for operation X_ConfigureWindow

Today I received the same error when clicking "file", or  "View" at the top menu, with the menu folding open but being blank. Strangely enough, when I drag the entire GUI window down (at approx 10% of the screens height, more or less) it doesn give this error, and the menu's open just fine.

PS. I also notice the error a few times more in the Thttps://github.com/open-ephys/plugin-GUI/issues/305erminal when launching the GUI.

Linux build on Ubuntu 16.0

reinderdorman, 2019-07-30:
edi: possibly related, downloaded latest version, and here the menu's don't open at all except for view. No error given in the Terminal.

jsiegle, 2022-07-30:
If you're still experiencing this problem with the latest version of the GUI, please re-open this issue.

closed 2022-07-30 by jsiegle