open-ephys/plugin-GUI #314: libokFrontPanel.so: file not recognized [Raspian-vJuly1029/Raspberry Pi Model 4]

timsainb, 2019-08-20:
I'm trying to build Open Ephys GUI on the new Raspberry Pi 4 from the [linux build from source tutorial](https://open-ephys.atlassian.net/wiki/spaces/OEW/pages/491546/Linux). Every step of the process works perfectly until building the plugins (`make -f Makefile.plugins`) where I get the following error:

```
Building RhythmNode.so
/home/pi/github/plugin-GUI/Builds/Linux/build/libokFrontPanel.so: file not recognized: file format not recognized
collect2: error: ld returned 1 exit status
make[1]: *** [Makefile:25: /home/pi/github/plugin-GUI/Builds/Linux/build/plugins/RhythmNode.so] Error 1
make[1]: Leaving directory '/home/pi/github/plugin-GUI/Source/Plugins/RhythmNode'
make: *** [Makefile.plugins:77: ../../Source/Plugins/RhythmNode] Error 2
```
Any idea why the file is not recognized, or how I can make the file recognizable?

aacuevas, 2019-08-20:
Although the raspberry pi uses Linux, its CPU is not a x86 (Intel/AMD) processor but an ARM. 
The frontpanel library we included, which we got from Opal Kelly, the FPGA makers, is built for x86 CPUs, so an ARM machine will not be able to load it.

That said, I noticed that Opal Kelly has very recently released ARM versions of their library. If you write me to aaron (at) open-ephys.org I could send you the file. Keep in mind that at this moment open ephys is developed for desktop pcs only. And while this development from Opal Kelly would allow us to make a Raspi version of the GUI, which could be interesting, it is not something we currently support, so there might be other unforeseen issues.

Best,
Aarón

timsainb, 2019-08-20:
Awesome, thank you, I will do that. By the way, I just posted on the google group without having seen this (https://groups.google.com/forum/#!topic/open-ephys/7IqALiLHJYc), so please ignore that post!

timsainb, 2019-08-21:
Update: @aacuevas sent me the ARM build of libokFrontPanel.so for the raspberry pi, then everything built fine. After that I just had to install the boost crypto library (i installed all of the boost libraries, unsure if that was needed `sudo apt-get install libboost1.62-*`). [Here's a video of it working](https://twitter.com/tim_sainburg/status/1164252464155021314).

closed 2019-08-21 by timsainb

aacuevas, 2019-11-27:
As an extra update, I have included the ARM library in the development branch, inside Resources/DLLs/Linux-ARM.

It will eventually move to the master branch in a future release.