open-ephys/plugin-GUI #322: Weird artifact/peak/spike every ~10s; on vers.>0.4.4.1; NOT on v.0.4.2.1

cskoven, 2019-10-07:
I recently pulled the newest source from this repo, and compiled it.
However I get a weird "artifact/peak/spike/something" on what appears to be all my ADC-channels - every 10s.

[image]

I've tried to use the latest released binary file, which give the same weird artifact.
My old version (compliled from source; v.0.4.2.1) does not give this weird artifact.

I've tried to setup the "pipeline"/signal chain completely similar - with/without using saved config files. But the problem seems to lie with the version of the software, rather than the signal chain config.

When compiling, I get a lot of lines (not errors per se) about "class LookAndFeel" (forward declaration), but it finishes the compilation and the program runs fine. I don't know if this has anything to do with it? I guess it shouldn't, as it's also a problem with the binary v.0.4.4.1-file.

It's running on a machine with: Ubuntu 18.04.2 LTS

Does anyone have any insight?

anjaldoshi, 2019-10-07:
Hi,

Thanks for providing a detailed analysis on the issue. We're aware of this "artifact", and it is a part of my TODO list.

I had one detail that I wanted to confirm with you. Did you get a chance to look at the recorded data and see whether this artifact shows up in it?

anjaldoshi, 2020-09-16:
Fixed in PR #348

closed 2020-09-16 by anjaldoshi