open-ephys/plugin-GUI #324: Recording, sample counts don't make sense

holmescharles, 2019-10-14:
I just installed Open-Ephys following the instructions on the wiki (build from source, Linux) and recorded 10 s of data. I am using the Binary version of the output. I am set to record from 32 channels at 30kHz. I would expect that I should have:

(10 s) x (30000 samples/s) x (2 bytes per sample) x (30 channels) = 18,000,000 bytes ~ 18 MB

Instead, the continuous file (continuous.dat) is only 19536 bytes, almost 1000x smaller than I'd expect.

Can you think of any reason this might be happening? I've reproduced this several times, so I am convinced it's not a fluke. Its either a bug or human-error on my part.

aacuevas, 2019-10-14:
That is strange indeed, your calculations are right.

Is the timestamps file also smaller than it should? Running the GUI from console, do you get any kind of error message? Could you please send me (to aaron (at) open-ephys.org) both the settings.xml and structure.oebin files related to those recordings so I could check for any possible explanation?

holmescharles, 2019-10-14:
Timestamps should be:
(30000 samples/s) * (10 s) * (8 bytes / sample) = 2,400,000 bytes.

But it is 2448  bytes instead.
I will email you those files you mentioned now.

holmescharles, 2019-11-04:
Found the problem. User error.

closed 2019-11-04 by holmescharles

tang37, 2019-12-29:
Hello,

I have the same problem. What use error did you have? I use a TTL signal to trigger a recording for 2s, but the flat binary data is either more or less than I expected. I am not sure if it is because of that.

Appreciate it for any help,
Yu