open-ephys/plugin-GUI #325: Eventbroadcaster missing

holmescharles, 2019-10-15:
I just updated OpenEphys to the newest version, and I cannot find the Event Broadcaster module. Has it been deprecated?

aacuevas, 2019-10-15:
I have checked the binary distributions and it should be there. Please confirm if it is present in your Plugins folder.
Also, if present, it might not be loading for some reason. If it were the case, try opening the GUI in a console and checking the console contents when loading the plugins to help diagnose the issue.

holmescharles, 2019-10-16:
I should have been more clear: I mean it seems to be missing from the github source code. I need to make a small change so that it only sends clustered spike events (not all events) in the C++ code.

aacuevas, 2019-10-16:
Ah, all plugins requiring external libraries have been moved to different [repositories](https://github.com/open-ephys-plugins/) so the core GUI can be compiled even if those libraries are not installed.

In this case, plugins requiring ZMQ have been moved to the [ZMQPlugins repository](https://github.com/open-ephys-plugins/ZMQPlugins)

holmescharles, 2019-10-29:
Got it! Thanks,

closed 2019-10-29 by holmescharles